stereolabs/zed-ros-wrapper #106: Update README

m-zain-khawaja, 2017-02-16:
Hello, 
I have the newest version of the ZED SDK (1.2.0) with ROS Kinetic. The topic names on your README are prefixed with camera/... however, they are actually listed as zed/...

Thank you!

shoemakerlevy9, 2017-03-13:
FYI: This is set in the launch file (see below). You could have also changed the group there to be 'camera' instead of 'zed'. But you are correct, since the default is now ZED the README should be updated.
```

  <group ns="zed">
    <include file="$(find zed_wrapper)/launch/zed_camera.launch">
      <arg name="svo_file" value="$(arg svo_file)" />
      <arg name="zed_id" value="$(arg zed_id)" />
    </include>
  </group>
```

GSORF, 2017-04-08:
This issue will be solved once PR #114 will be merged into master.

GSORF, 2017-04-10:
The issue is solved now. @m-zain-khawaja can we close this issue?

closed 2017-04-10 by m-zain-khawaja