stereolabs/zed-ros-wrapper #108: -lopencv_dep_cudart issue

griz1112, 2017-02-17:
Working on zed_ros_wrapper 1.2.0 with 1.2 sdk on arm64. Ran into the -lopencv_dep_cudart issue. If you go into CMakeLists.txt starting at line 23 you will notice that some of the set commands are lower case. They are supposed to be all upper case. So change any set to SET and it works fine.

GSORF, 2017-02-27:
Hi griz1112,

I can confirm this issue on the NVIDIA Jetson TX1 (arm64, Ubuntu 16.04), zed_ros_wrapper version 1.2.0 with SDK version 1.2.

According to the CMAKE documentation this should be inn upper case like you mentioned (compare https://cmake.org/Wiki/CMake/Language_Syntax#SET_and_substitution).

I will make a new PullRequest to fix your issue.

griz1112, 2017-02-27:
Sorry I'm new with all this so I don't know how to do all that yet. 
Still doesn't always work. Make sure you look at the versions and get 
the one that matches your sdk version. Also to get rosdep to install all 
the deps you need to start it with rosdep install zed_wrapper 
--skip_keys="opencv" Also to get openni support it needs to be installed 
before you build zed_wrapper. If it comes up with the libcudart error 
source devel/setup.bash that usually fixes it. Its a strange error. I've 
been all over it and still haven't pinned it down totally. Works though 
and you only have to build it once :)

Dan

GSORF, 2017-02-28:
Hi Dan,

Ok, thanks for your reply. In my case it fixed the error without any further modifications. And since the SET command was clearly wrong according to the CMAKE docs, I opened the PR which was already merged into the master branch a few hours ago (Thanks @adujardin :) ).

So at least we have one bug less now :+1: 

I still need to find out why my Jetson TX-1 stops publishing point clouds from the ZED after 10 - 20 seconds when viewed in RViz, though...

Best regards,
Adam

griz1112, 2017-02-28:
I will see if mine is doing that next time i have it up and running. I 
don't think it does. I'be been running rtabmap with it for a slam 
system. Yesterday I finally was able to build orb_slam. Its nice. I need 
to figure out how to use the Zed as an openni camera. Do I just load the 
driver and set up the ID in the openni init files? I know about the 
startup parameter but didn't figure that turned it on as far as openni 
is concerned.

Dan

GSORF, 2017-04-10:
Not really sure how to use it with OpenNI, but for OpenCV it is sufficient to connect the ZED via USB and open the first VideoDevice to capture from it. For point clouds I would still use the ZED wrapper provided here. If you want low level access you can look up the code in the Wrapper to get information about how to access the hardware and use the API provided by the ZED SDK.

Is this issue closed now? Could you close it, please?

Kind regards,
Adam

closed 2018-06-28 by Myzhar