stereolabs/zed-ros-wrapper #112: nothing show in camerainfo

yxftju, 2017-02-24:
Hi , I try zed-ros-wrapper just now but I can get nothing with /zed/left/camerinfo , /zed/right/camerainfo has nothing either. I want to check if there is a bug . 

I check the info using `rostopic echo` command.

skylvr1289, 2017-03-02:
I have the same problem ? No camera_info topic is publishing anything?

AAdelBayoumi, 2017-03-02:
same problem here also, is there any way to solve this issue ?

ajayfuloria, 2017-03-03:
I am facing the same problem, my setup is Ubuntu 16.04, CUDA 8, ZED SDK 1.2, ROS Kinetic. When I try to echo any of the camera_info topic , nothing is being published. Please help if anyone has a solution.

skylvr1289, 2017-03-03:
I noticed that, whenever some node in ROS is using (subscribed & running) , at that time, if you echo camera_info, the data will be shown. But as soon you stop the node, no data will be echoed..

yxftju, 2017-03-04:
I subscribe the topic but nothing works. It seems that that topic should be publish by a service but not a publisher.

PingHGao, 2017-05-12:
You can get the camera info when you are reading the image. This means the camera info topic is empty when you are only reading the topic.

ndspike, 2017-05-18:
I think my group has experienced a similar problem. We have Zed mounted on a mobile robot. After launching the zed wrapper, we can echo all Zed topics when connected via ssh, but through ROS master URI, we cannot echo camera topics. (other topics not associated with Zed echo as normal such as /scan, /imu_data, etc) 

This has prevented us from viewing odometry and point cloud information using rviz.

Does this sound like the issue people are experiencing?

akashjinandra, 2018-03-13:
Anybody make any progress on this?

jediofgever, 2018-04-01:
@PingHGao  I dont understand. What do you mean reading image ?. I would like to subscribe this topic to recieve messages. Can you be more specific

fabolhak, 2018-04-17:
@jediofgever you have to subscribe to one of the image topics (for right camera: `/right/image_raw_color` or `/right/image_rect_color`) e.g. using the ROS image viewer. Only then the camera_info message of the respective camera is also being published.

shiveshkhaitan, 2018-04-19:
@jediofgever I have subscribed to the image topic /rgb/image_raw, the topic in which the image is being published. Still there is nothing published on /rgb/camera_info topic.

shiveshkhaitan, 2018-04-19:
@PingHGao @fabolhak I have subscribed to the image topic in rviz. Still there is nothing being published on rgb/camera_info topic. Is there anything else apart from subscribing to image to get the info published?

shiveshkhaitan, 2018-05-01:
ZED plugin doesn't publish camerainfo. But when tried with actual ZED camera it was getting published.

Myzhar, 2018-06-28:
@fabolhak correctly replied to the issue

closed 2018-06-28 by Myzhar