stereolabs/zed-ros-wrapper #115: Fail to load libraries libopencv_*.so.3.1.

ChrisPsa, 2017-03-08:
ROS kinetic updated opencv to version 3.2 and zed_ros_wrapper tries to find .so.3.1, resulting in an error and fail to load nodelet.

temporary fix is to create links 

`libopencv_*.so.3.1 -> lilbopencv_*.so.3.2`

Butakus, 2017-04-18:
I have the same issue. I tried to compiled the package with OpenCV 3.2 instead of 3.1 changing the CMakeLists.txt from:

`find_package(OpenCV ${OCV_VERSION} COMPONENTS core highgui imgproc REQUIRED)`
to:
`find_package(OpenCV 3.2 COMPONENTS core highgui imgproc REQUIRED)`
or:
`find_package(OpenCV 3 COMPONENTS core highgui imgproc REQUIRED)`

but the error still ocurrs, so I guess that the issue is not related to the ros wrapper, but to the SDK itself... Am I wrong?

Does the SDK only works with OpenCV 3.1?
Is it possible to have a fix for this?

Thanks.

maria-miranda, 2017-05-23:
SDK is compiled with OpenCV 3.1 so there is no way to change the OpenCV version on CMake.

Would be great if ZED team @nesnes compile an SDK version with OpenCV 3.2 so we could only use one OpenCV version :)

adujardin, 2017-06-02:
Hi,

This issue will be solved in the upcoming release since we removed the OpenCV dependency from the ZED SDK. You'll be able to link the ROS wrapper with any OpenCV version.

Jackzhemliu, 2017-06-14:
When will be the new release?

KarimHassanieh, 2017-07-16:
Im having this issue can anyone help in this matter??
its really frustrating as i dont know how to create the link

adujardin, 2017-07-28:
Hi,
Just to inform you that the new release of the ZED SDK (which remove OpenCV dependency) is out! https://www.stereolabs.com/developers/release/2.1/

closed 2017-10-16 by adujardin

puren, 2017-11-14:
Hi,

I tried both 2.2 and 2.1 release but I still get "error while loading shared libraries: libopencv_highgui.so.3.1:" when I run ./Zed_Camera_Control.
I have ubuntu 16.04, opencv 3.3,  cuda 8.0.

Were others able to solve the opencv issue with the new releases?