stereolabs/zed-ros-wrapper #116: cannot launch node of type...

ahsteven, 2017-03-10:
I am trying to run the ros-wrapper for the first time and I get an error after running:

$ roslaunch zed_wrapper zed.launch

The error is:

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /zed/zed_description: <?xml version="1....
 * /zed/zed_wrapper_node/depth_cam_info_topic: depth/camera_info
 * /zed/zed_wrapper_node/depth_topic: depth/depth_regis...
 * /zed/zed_wrapper_node/frame_rate: 60
 * /zed/zed_wrapper_node/gpu_id: -1
 * /zed/zed_wrapper_node/left_cam_info_topic: left/camera_info
 * /zed/zed_wrapper_node/left_raw_topic: left/image_raw_color
 * /zed/zed_wrapper_node/left_topic: left/image_rect_c...
 * /zed/zed_wrapper_node/odometry_db: 
 * /zed/zed_wrapper_node/odometry_topic: odom
 * /zed/zed_wrapper_node/openni_depth_mode: 0
 * /zed/zed_wrapper_node/point_cloud_topic: point_cloud/cloud...
 * /zed/zed_wrapper_node/quality: 1
 * /zed/zed_wrapper_node/resolution: 2
 * /zed/zed_wrapper_node/rgb_cam_info_topic: rgb/camera_info
 * /zed/zed_wrapper_node/rgb_raw_topic: rgb/image_raw_color
 * /zed/zed_wrapper_node/rgb_topic: rgb/image_rect_color
 * /zed/zed_wrapper_node/right_cam_info_topic: right/camera_info
 * /zed/zed_wrapper_node/right_raw_topic: right/image_raw_c...
 * /zed/zed_wrapper_node/right_topic: right/image_rect_...
 * /zed/zed_wrapper_node/sensing_mode: 1
 * /zed/zed_wrapper_node/svo_filepath: 
 * /zed/zed_wrapper_node/zed_id: 0

NODES
  /zed/
    zed_base_link1 (tf2_ros/static_transform_publisher)
    zed_base_link3 (tf2_ros/static_transform_publisher)
    zed_state_publisher (robot_state_publisher/state_publisher)
    zed_wrapper_node (zed_wrapper/zed_wrapper_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[zed/zed_base_link1-1]: started with pid [14330]
process[zed/zed_base_link3-2]: started with pid [14331]
ERROR: cannot launch node of type [zed_wrapper/zed_wrapper_node]: can't locate node [zed_wrapper_node] in package [zed_wrapper]
process[zed/zed_state_publisher-4]: started with pid [14340]

shoemakerlevy9, 2017-03-13:
Try "rospack find [package]". If that returns an error you may need to source the /devel/setup.bash file of your catkin workspace

ndjaitly, 2017-07-10:
I had a similar issue. And rospack did indeed find the file. Any other pointers ?

ndjaitly, 2017-07-10:
Silly -- just forgot to do the obvious:
$catkin_make
$source ./devel/setup.bash

closed 2018-06-28 by Myzhar

venbalas, 2019-03-07:
I m facing the same issue.. sourcing devel setup, catkin clean build nothing works.. Not sure where the packages are copied..

philwalker2, 2019-06-18:
Hello All, 
I tried the solution recommended that resulted in the closing of this topic and it did not work. 

I have the following error message. 

RLException: [display.launch] is neither a launch file in package [zed_display_rviz] nor is [zed_display_rviz] a launch file name Z
The traceback for the exception was written to the log file

catkin_make ran successfully. I sourced it as recommended. 

Thanks

Myzhar, 2019-06-18:
Hi @StairWalkerOne,
If you are using the latest version of the wrapper you must launch `display_zed` or `display_zedm`:
https://www.stereolabs.com/docs/ros/#displaying-zed-data

Daneduke7, 2020-09-28:
I received the same error, but I was wondering if it had anything to do with the fact that I don't have a zed camera operating at the same time I launch the software?

DanielSepeda, 2022-03-11:
I had the same issue, but the solution was to do _catkin_make install_ if anyone is having similar issues

Myzhar, 2022-03-12:
@DanielSepeda if you need to install the ROS wrapper it means that you have not correctly configured the path to your catkin workspace. 
Add this line at the end of your `.bashrc` file, replacing ` <catkin_ws>` with the full path of your catkin workspace
`source <catkin_ws>/devel/setup.bash`