stereolabs/zed-ros-wrapper #119: Setting confidence

Galto2000, 2017-03-27:
Dear sir or madam,

I am inquiring about the confidence value, which appears to be hard-coded on line 681 (at the time of writing this) to the value 80  (confidence = 80; )

Is there any reason why this value cannot be passed as a parameter, or am I missing something?

Thanks in advance!

Galto

HardTaskToFindAName, 2017-05-04:
Hey Stereolabs Team,

I would like to ask the same question in our documentation 
is stated that it should be a parameter that can be configured 
with ROS dynamic reconfigure but this is not the case. 

I also want to ask how I can achieve the same depth map 
which I get with the DepthViewer from the SDK. So which 
parameter setup is equivalent the one of the DepthViewer.

Best

Manuel

nesnes, 2017-05-05:
Hi,
I've just updated the wrapper to fix the dynamic reconfigure. 2628331 
I've also added the possibility of configuring it from the launch file. c7924d92

Thank you for pointing out this issue.

pavloblindnology, 2017-06-15:
Hi,
1) Is there a way to get variance or st.dev. measure from ZED camera?
2) What invalidity flag is used for disparity\depth measure? I see in my results both NaN and Infinity values.
Thanks.

closed 2018-06-28 by Myzhar