stereolabs/zed-ros-wrapper #121: Variance on tracking

bcrobo, 2017-05-06:
Hi All,

First I want to mention this is not a BUG ticket. It's more an open discussion in order to get some precision on output data of the ZED camera.

I'm currently using the ZED Camera in a personnal project on sensor fusion. As it is common in such algorithms, I'm using a probabilistic framework to estimate the best estimate of my current pose among all my sensors. I haven't merged yet the ZED camera in it for one major reason : I don't have any variance information of the tracking state of the ZED camera.

I'm using ROS and I looked at the covariance matrix fields of the /zed/odom topic and I saw the variances are not set at all. I looked deeper in the ZED sdk where I hoped to find a parameter to enable in order to get such information. Without success... 

Here is my questions : 
- Is variances of the tracking state available somewhere through the ZED sdk ?
- If no, is it in the pipe to release a sdk version with it in the next months ?
- If, no how could I estimate such variances depending of the knowledge of your system ? (I'm aware this question is a big open topic since it probably depend on your internal block-matching and tracking algorithm but maybe you could provide me some tips on which direction to take).

Thank you,

adujardin, 2017-05-09:
Hi,

The variances of the tracking are not yet available. However, we are currently working on it and it should be included in the next few months (probably not the upcoming release but the next one).
We will update the ROS package accordingly once it's released.

In the meantime, I don't think you can really estimate the variances from the current SDK. You could have a guess of the precision from the tracking confidence (`sl::Pose::pose_confidence` [API doc](https://www.stereolabs.com/developers/documentation/API/classsl_1_1Pose.html))  but I'm afraid it's quite far from it, as it's just an integer factor scaled empirically.

bcrobo, 2017-05-10:
Hi adujardin,

Ok good to hear ! I'm also not sure `sl::Pose::pose_confidence` is good enough. I'm looking forward the next releases.

Thank you for your answer.

pavloblindnology, 2017-06-19:
Hi,

Do I understand right that position tracking can be computed ONLY if depth map is computed too? 
I.e., I cannot use ZED position tracking without using ZED depth computation?

Thanks.

adujardin, 2017-06-19:
Hi @pavloblindnology,

Yes, that's correct.

pavloblindnology, 2017-06-19:
Do you plan to separate those features or position tracking uses some depth info to proceed?

adujardin, 2017-06-19:
It depends on the depth info to compute the position, so it's not possible

micander, 2017-11-13:
Is there an ETA on this issue?

natejgardner, 2017-11-13:
Any updates on this? When will pose and twist covariance be available? Due to latency in image_transport, it is very preferable to calculate odometry using the sdk. However, the odometry data is not very useful without covariance.

davidbsp, 2018-02-16:
This enhancement would be really welcome.

reliassen, 2018-02-21:
Looking forward to this as well

tyronow, 2018-05-29:
This would be great!

closed 2018-11-12 by Myzhar

davidbsp, 2018-11-12:
Hi. Was this solved?

Myzhar, 2018-11-12:
Yes, with the latest version the covariance has been added to Odometry and to Pose with the topic pose_with_covariance

davidbsp, 2018-11-12:
Great! Thank you!