stereolabs/zed-ros-wrapper #124: zed-wrapper-node running on TX1 freezes when point clouds are subscribed to over LAN

thdieterle, 2017-05-23:
Hello guys,

I am using the wrapper on a Jetson TX1 and I want to visualize the point cloud in RViz on another computer connected via LAN.
When I subscribe to the point cloud topic, the zed-wrapper-node always freezes after running for a couple of seconds up to a few minutes. The node becomes completely unresponsive and has to be killed. 

Anyone experienced a similar issue that he could solve? Thanks.
Regards, 
Tom

maciejmatuszak, 2017-06-16:
Thanks for the hint on the LAN, yes I am experiencing the same issue. Subscribing to point cloud over LAN will freeze it. Subscribing on local machine seems to have no problem. 
I put several debug statement in the wrapper and it always stuck on the retrieveImage:
`    NODELET_DEBUG("Retrieved Left rect image...");
    zed->retrieveImage(leftZEDMat, sl::VIEW_LEFT);
    NODELET_DEBUG("Retrieved Left rect image...done");`
[DEBUG] [1497587696.813698003]: Loop Start
[DEBUG] [1497587696.813959250]: Loop will run
[DEBUG] [1497587696.814101176]: grab..
[DEBUG] [1497587696.826604493]: grab..DONE
[DEBUG] [1497587696.826831158]: Retrieved Left rect image...
[DEBUG] [1497587696.911440293]: Retrieved Left rect image...done
[DEBUG] [1497587696.912858144]: Retrieved Left image...
[DEBUG] [1497587696.915280985]: Retrieved Left image...done
[DEBUG] [1497587696.917305237]: Retrieved Right rect image...
[DEBUG] [1497587696.919942711]: Retrieved Right rect image...done
[DEBUG] [1497587696.923325127]: Retrieved Right image...
[DEBUG] [1497587696.925500054]: Retrieved Right image...done
[DEBUG] [1497587696.947024951]: Loop Start
[DEBUG] [1497587696.947211304]: Loop will run
[DEBUG] [1497587696.947337292]: grab..
[DEBUG] [1497587696.965055924]: grab..DONE
[DEBUG] [1497587696.965268734]: Retrieved Left rect image...
[DEBUG] [1497587696.984288135]: Retrieved Left rect image...done
[DEBUG] [1497587696.986919412]: Retrieved Left image...
[DEBUG] [1497587696.990900207]: Retrieved Left image...done
[DEBUG] [1497587696.993452578]: Retrieved Right rect image...
[DEBUG] [1497587696.995732817]: Retrieved Right rect image...done
[DEBUG] [1497587696.997408530]: Retrieved Right image...
[DEBUG] [1497587696.999870121]: Retrieved Right image...done
[DEBUG] [1497587697.027020901]: Loop Start
[DEBUG] [1497587697.027506157]: Loop will run
[DEBUG] [1497587697.027823394]: grab..
[DEBUG] [1497587697.039663176]: grab..DONE
[DEBUG] [1497587697.040014995]: Retrieved Left rect image...
^C[zed/zed_state_publisher-5] killing on exit
[zed/zed_base_link4-3] killing on exit
[zed/zed_base_link3-2] killing on exit
[zed/zed_wrapper_node-4] killing on exit
[zed/zed_base_link1-1] killing on exit
^[[1;2A
[zed/zed_wrapper_node-4] escalating to SIGTERM
shutting down processing monitor...
... shutting down processing monitor complete
done
ubuntu@tegra-ubuntu:/data/git/miner-tx1/catkin_ws$

I reinstalled cuda, opencv and ZED SDK but still the same problem. 
@thdieterle did you manage to fix it?

Maciej

maciejmatuszak, 2017-06-16:
Out of curiosity I did the opposite, run the zed_wrapper on laptop and subscribed to the point cloud on the tx1 and it worked ok. It seems to be specific to something Tx1, maybe the opencv 2.4?
I am running ubuntu 16.4 (Jetpack 3.0) with ZED SDK 2.0.1 and zed wrapper with master as of today.

Maciej

thdieterle, 2017-06-21:
Hi Maciej,
Unfortunately, I was not able to solve the problem. I managed to finish my project using the TK1. Here, the node does not freeze. The issue really seems to be specific to the TX1.  
 
Thomas

Myzhar, 2018-07-02:
@thdieterle it's been a while since you opened this issue. I tried to replicate it on Jetson TX1, but everything seems right.
Have you solved it on your machine with the latest updates?

thdieterle, 2018-07-03:
@Myzhar I can't tell, since I don't have access to the hardware anymore. Sorry
Best regards, 
Thomas

closed 2018-11-24 by Myzhar