stereolabs/zed-ros-wrapper #127: High-variance depth errors for flat surfaces

ana-GT, 2017-06-01:
Hi. I intend to use our ZED camera for robot perception (passing pointclouds to an octomap in order to perform collision detection). I am observing that the pointcloud output of the ZED camera is highly variant, for surfaces such as walls. In the picture attached, you can see a pointcloud obtained with the ZED camera and the picture of the scene to the left: The wall behind the robot arm appears highly curved. Is this behavior normal, and if not, could you kindly recommend an approach to reduce this error? The picture shown was obtained by using NO filtering (I attempted to use different confidence values and the results were far worse). 

Thanks!

[image]

adujardin, 2017-06-02:
Hi,

You can change the depth mode [parameter](https://github.com/stereolabs/zed-ros-wrapper#launch-file-parameters) in the launch file. Update [this line](https://github.com/stereolabs/zed-ros-wrapper/blob/master/launch/zed_camera.launch#L15) and put the value `3` to get the best available depth quality, a GTX 770 will easily handle it.

In the upcoming release (ZED SDK 2.1), we added a depth stabilizer mode which significantly improves the precision and the stability of the depth map (it uses the positional tracking). We will update the ROS wrapper accordingly once it's available.

ana-GT, 2017-06-02:
Hi @adujardin,

Thanks for the suggestions. I changed the quality parameter and the results are rather similar (there appear a few extra holes with the higher quality parameter, though).

Is there any estimated date for the release of the 2.1 ZED SDK? I am trying to figure out if I should try to improve the results with some filtering on my own in the left/right pairs or if it would be just better to wait for the next release.

Thanks!

adujardin, 2017-06-02:
The official release should be within a couple of weeks.

You can contact the support at support@stereolabs.com to get a beta version of the 2.1. In that case, you should mention this issue to get a pre-version of the ROS wrapper.

ana-GT, 2017-06-02:
@adujardin, Thank you very much! Just sent an email. Looking forward to try the beta version if possible.

closed 2018-06-28 by Myzhar

chaitanyantr, 2019-09-05:
hi @ana-GT ,

May i know what is the launch file for your 3d-map in octomap_server.

I'm trying to create a octomap with zed,bit bewildered what inputs from zed camera i have to give to octomap to create the map.

my_launch_file is...

<launch>
	<node pkg="octomap_server" type="octomap_server_node" name="octomap_server">
		<param name="resolution" value="0.05" />
		
		
		<param name="frame_id" type="string" value="/zed/zed_node/pose" />
		
		
		<param name="sensor_model/max_range" value="5.0" />
		
		
		<remap from="cloud_in" to="/zed/zed_node/point_cloud/cloud_registered" />
	
	</node>
</launch>