stereolabs/zed-ros-wrapper #130: ROS Multi Cam Error code: Low USB Bandwidth

adurandp, 2017-06-09:
Hi,
I tried to run two ZED cameras with VGA@15 using either the ROS wrapper (multi_cam and multi_gpu) or zed-multi-input.  In both cases, I have the following error for the second camera: "Error code: Low USB Bandwidth". I tried several configurations: on two USB 3.0 ports, one USB 3.0 port, and one Thunderbolt. It always gives me the same type of error.
My config: Dell Precision 7510, Intel® Core™ i7-6820HQ CPU @ 2.70GHz × 8 , Quadro M2000M, Ubuntu 16, ROS Kinetic.
By the way, I don't have any issue running a single camera either with ROS or the OpenCV example.

Any clue about this issue?
Thank you,
Adrien

Here is the log:
[ INFO] [1497052357.866528136]: Initializing nodelet with 8 worker threads.
[ INFO] [1497052357.886079689]: Initializing nodelet with 8 worker threads.
ZED (Init) >> Depth mode: Performance
ZED (Init) >> Depth mode: Performance
ZED (Init) >> Video mode: VGA@15
ZED (Init) >> Video mode: VGA@15
[ INFO] [1497052360.965537387]: Error code:  Success
[ INFO] [1497052362.974525539]: Reconfigure confidence : 100
[ INFO] [1497052363.161878119]: Advertized on topic rgb/image_rect_color
[ INFO] [1497052363.179955093]: Advertized on topic rgb/image_raw_color
[ INFO] [1497052363.198598990]: Advertized on topic left/image_rect_color
[ INFO] [1497052363.215578554]: Advertized on topic left/image_raw_color
[ INFO] [1497052363.233502876]: Advertized on topic right/image_rect_color
[ INFO] [1497052363.251627590]: Advertized on topic right/image_raw_color
[ INFO] [1497052363.270178610]: Advertized on topic depth/depth_registered
[ INFO] [1497052363.270712410]: Advertized on topic point_cloud/cloud_registered
[ INFO] [1497052363.271242150]: Advertized on topic rgb/camera_info
[ INFO] [1497052363.271742612]: Advertized on topic left/camera_info
[ INFO] [1497052363.272271013]: Advertized on topic right/camera_info
[ INFO] [1497052363.272795759]: Advertized on topic depth/camera_info
[ INFO] [1497052363.273286897]: Advertized on topic odom
ZED (Init) >> WARNING: Corrupted frames detected. Check that your ZED is connected to a USB3.0 port
[ INFO] [1497052373.716608214]: Error code: Low USB Bandwidth
ZED (Init) >> Depth mode: Performance
ZED (Init) >> Video mode: VGA@15

adurandp, 2017-06-15:
Hi
To update about this issue I would like to let you know I was able to run tree cameras using the sdk opencv example. It seems then it has to be an issue with the ROS-wrapper.

ravikt, 2018-08-03:
Hi @adurandp , Try to use a USB 3.0 powered hub to connect your cameras.

Myzhar, 2018-09-08:
I close the issue but you can continue to pay comments if you need further help

closed 2018-09-08 by Myzhar