stereolabs/zed-ros-wrapper #131: Failed to load nodelet [/zed/zed_wrapper_node]

alexge233, 2017-06-20:
Hi,

This is a vanilla Ubuntu 16.04 with ros-kinetic and a Titan GTX using CUDA 8.0.
I've manually built and installed opencv-3.2, and I am following the instructions from: http://wiki.ros.org/zed-ros-wrapper

The problem is after I've built the ros wrapper, I try to launch it, with the following error:

```bash
[ERROR] [1497980991.229978351]: Failed to load nodelet [/zed/zed_wrapper_node] of type [zed_wrapper/ZEDWrapperNodelet] even after refreshing the cache: Failed to load library /home/alex/catkin/devel/lib//libZEDWrapper.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libopencv_videoio.so.3.1: cannot open shared object file: No such file or directory)
[ERROR] [1497980991.230041390]: The error before refreshing the cache was: Failed to load library /home/alex/catkin/devel/lib//libZEDWrapper.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libopencv_videoio.so.3.1: cannot open shared object file: No such file or directory)
```

The library version appears to be hardcoded to 3.1? I've followed a now closed issue on Github which used symbolic links from 3.1 to 3.2 but each time I create one, another one becomes an issue.
Is there a fix for this?

tngan, 2018-03-11:
@alexge233 I have the same issue with jetson TK1. Have you resolved the problem ?

Russ76, 2018-03-14:
I am getting this error also. Tried to install Opencv from source and I'm not sore if that went as planned, and the error still comes up. This is with Xenial and the Zed plugged right into my Acer laptop. The Zed works perfectly with Rviz. Now I'm trying to get it to work with Rtabmap.

alexge233, 2018-03-14:
@tngan and @Russ76 sorry guys I never managed to solve the error and just gave up. My guess is someone has hardcoded a library version somewhere...

tngan, 2018-03-14:
@alexge233 Never mind, at least it works in older version, I have another TK1 which is running 14.04, SDK v1.2 and wrapper with release 1.2.0 (https://github.com/stereolabs/zed-ros-wrapper/tree/v1.2.0).

closed 2018-06-28 by Myzhar