stereolabs/zed-ros-wrapper #132: Problem accessing compressed Images

heimsi, 2017-06-22:
Hello,

I have the following setup: My ZED cam is attached to a Jetson TX1, where a _zed_wrapper_node_ runs. It is connected to a ROS master via LAN. I am able to look at the images and the depth map produced by the ZED cam on the second computer running the ROS Master without problems.

In the long term the setup should work over WLAN, that's why I would like to use the compressed image transport. So on the ROS master computer I launch a republish node which should decompress the compressed images of the left ZED camera:
```
<launch>
 <node name="republish_image_rect_color_left" type="republish" pkg="image_transport" output="screen" 
    args="compressed in:=/zed/left/image_rect_color raw out:=/zed/republish/left/image_rect_color"
    />
</launch>
```
As soon as I launch this file, the _zed_wrapper_node_ crashes with the following error:

`[zed/zed_wrapper_node-4] process has died [pid 3892, exit code -11, cmd /home/ubuntu/catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node __name:=zed_wrapper_node __log:=/home/ubuntu/.ros/log/bc6a3bb0-572f-11e7-824f-705ab6919432/zed-zed_wrapper_node-4.log].`

When I launch the zed_wrapper_node in debug mode with  `launch-prefix="gdb -ex run --args" ` in zed_camera.launch, debugging shows the following:

```
[ INFO] [1498125769.037238963]: Advertized on topic rgb/image_rect_color
[ INFO] [1498125769.222155749]: Advertized on topic rgb/image_raw_color
[ INFO] [1498125769.419520434]: Advertized on topic left/image_rect_color
[ INFO] [1498125769.618201623]: Advertized on topic left/image_raw_color
[ INFO] [1498125769.811763413]: Advertized on topic right/image_rect_color
[ INFO] [1498125770.007692116]: Advertized on topic right/image_raw_color
[ INFO] [1498125770.191289743]: Advertized on topic depth/depth_registered
[ INFO] [1498125770.197831581]: Advertized on topic point_cloud/cloud_registered
[ INFO] [1498125770.203473321]: Advertized on topic rgb/camera_info
[ INFO] [1498125770.209933962]: Advertized on topic left/camera_info
[ INFO] [1498125770.215496014]: Advertized on topic right/camera_info
[ INFO] [1498125770.220358487]: Advertized on topic depth/camera_info
[ INFO] [1498125770.226693713]: Advertized on topic odom
[New Thread 0x7f87bac190 (LWP 4167)]
[New Thread 0x7f86fac190 (LWP 4168)]
[New Thread 0x7f873ac190 (LWP 4169)]
[New Thread 0x7f8c486190 (LWP 4170)]

Thread 19 "zed_wrapper_nod" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7f87bac190 (LWP 4167)]
strlen () at ../sysdeps/aarch64/strlen.S:92
92	../sysdeps/aarch64/strlen.S: No such file or directory.
```

- Has anybody experience with this setup?
- Has anybody an idea what the error could be related to?

heimsi, 2017-06-22:
If I try to look directly at the compressed images with image_viewer:
```
rosrun image_view image_view image:=/zed/left/image_rect_color _image_transport:=compressed
```
I get the following error output for the zed_wrapper_node:

```
[ INFO] [1498128123.547980668]: Initializing nodelet with 4 worker threads.
ZED (Init) >> Depth mode: Performance
ZED (Init) >> Video mode: VGA@15
[ INFO] [1498128127.183010991]: Error code:  Success
[ INFO] [1498128129.221603634]: Reconfigure confidence : 100
[ INFO] [1498128129.964170020]: Advertized on topic rgb/image_rect_color
[ INFO] [1498128130.149290252]: Advertized on topic rgb/image_raw_color
[ INFO] [1498128130.329264001]: Advertized on topic left/image_rect_color
[ INFO] [1498128130.522703302]: Advertized on topic left/image_raw_color
[ INFO] [1498128130.712989704]: Advertized on topic right/image_rect_color
[ INFO] [1498128130.911181062]: Advertized on topic right/image_raw_color
[ INFO] [1498128131.109490180]: Advertized on topic depth/depth_registered
[ INFO] [1498128131.114838223]: Advertized on topic point_cloud/cloud_registered
[ INFO] [1498128131.120427776]: Advertized on topic rgb/camera_info
[ INFO] [1498128131.126864355]: Advertized on topic left/camera_info
[ INFO] [1498128131.133361767]: Advertized on topic right/camera_info
[ INFO] [1498128131.139451525]: Advertized on topic depth/camera_info
[ INFO] [1498128131.145859094]: Advertized on topic odom
[ERROR] [1498128884.304142140]: Compressed Image Transport - JPEG compression requires 8/16-bit color format (input format is: 32FC1)
[ERROR] [1498128884.355860553]: Compressed Image Transport - JPEG compression requires 8/16-bit color format (input format is: 32FC1)
[ERROR] [1498128885.381078457]: Compressed Image Transport - JPEG compression requires 8/16-bit color format (input format is: 32FC1)

[...]

[ERROR] [1498128891.899944953]: Compressed Image Transport - JPEG compression requires 8/16-bit color format (input format is: 32FC1)

[zed/zed_wrapper_node-4] process has died [pid 5075, exit code -11, cmd /home/ubuntu/catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node __name:=zed_wrapper_node __log:=/home/ubuntu/.ros/log/b5a5750e-5731-11e7-824f-705ab6919432/zed-zed_wrapper_node-4.log].
log file: /home/ubuntu/.ros/log/b5a5750e-5731-11e7-824f-705ab6919432/zed-zed_wrapper_node-4*.log
```

If I replace the ZED camera with another camera, and use the cv_camera node to publish the images, looking directly at the compressed image with the above command works. Also the setup in the previous post with the republish node works.

heimsi, 2017-06-23:
While I didn't get any further finding the problem, I found a work-around: have on the Jetson beside the _zed_ros_wrapper_node_ a [republish node](http://wiki.ros.org/image_transport#republish), compressing the raw images. But of course it would make the setup simpler to directly be able to use the ZED's compressed image topics.

rbonghi, 2017-07-25:
Hi @heimsi,

I have your same problem with a Jetson TX2 and Jetpack 3.0
When I try to open a compressed topic (left, right, rgb, depth) the zed wrapper node go instantaneously in crash:

```
[zed/zed_wrapper_node-2] process has died [pid 5527, exit code -11, cmd
/home/nvidia/catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node __name:=zed_wrapper_node 
__log:=/home/nvidia/.ros/log/e7dd5634-7155-11e7-b022-00044b80e856/zed-zed_wrapper_node-2.log].
log file: /home/nvidia/.ros/log/e7dd5634-7155-11e7-b022-00044b80e856/zed-zed_wrapper_node-2*.log
```
My launch file is available on: https://github.com/rbonghi/panther/blob/master/panther_bringup/launch/include/zed_stereo.launch

rbonghi, 2017-07-28:
Update
With the new zed_wrapper and SDK 2.1 the compressed images are not available.

andre-nguyen, 2017-09-21:
Why was compressed image support removed? Recording things raw is extremely impractical.

andre-nguyen, 2017-09-24:
@rbonghi @heimsi @nesnes I got it fixed on my end by installing the compressed image transport plugins and adding them as dependencies to the package.

chan-e91, 2017-10-20:
HI Heimsi,
"In my case the zed_ros_wrapper runs on a Jetson TX1, with VGA resolution and 15 fps. I can look at the raw image topics and raw depth topics without problems."

How and where we could change the resolution and fps while using ROS wrapper.
I am getting low bandwidth errorwhile using ZED with T

ofsource, 2020-05-07:
> @rbonghi @heimsi @nesnes I got it fixed on my end by installing the compressed image transport plugins and adding them as dependencies to the package.

Hi, andre-nguyen, how did you add dependencies to the package? I met the same problem, and I add_dependencies in both CMakeLists.txt and package.xml, though it still  crashed when the  compressed image topic was subscribed. Could you give some help?

Myzhar, 2020-05-07:
@ofsource 
`$ sudo apt install ros-melodic-image-transport-plugins`
I will update the package to add the dependency

Myzhar, 2020-05-07:
Fixed in https://github.com/stereolabs/zed-ros-wrapper/commit/331e343a2de66f5fbe586a0ffb0be79a59b97299