stereolabs/zed-ros-wrapper #133: zed sdk 2.0 zedfu

lioyijie, 2017-06-23:
Why do I have bad pointclouds from Zed Depth Viewer.
For ZEDFU, when I Select live input and press start, the mesh always hang there while the camera is still capturing images

heimsi, 2017-06-23:
To help you I think more information is needed. 
When you open the Zed Depth Viewer, what you see looks fine?
Calculating the mesh requires quite some computing power, so I guess it would be not unusual that the mesh lags a bit. When you move the camera, does the mesh move? Does it move if you wait let's say for 2 minutes? what if you stop capturing and press the "process" button?

lioyijie, 2017-06-27:
@heimsi when I double clicked the Zedfu, everything looks fine, but when i clicked on live and press start, the camera still continues capturing images but the mesh just hang there and stop moving, and on the right side of the Zedfu, theres no image or anything displayed on the right side. 
However, if I select the input as svo and select the videos that I recorded before, there will be image forming out slowly, but the output is bad and does not look like anything from the real image

heimsi, 2017-06-27:
hmm... maybe your computer is unable to cope with capturing the images and doing the point cloud fusion at the same time? At which workload is your computer when running zedfu?

lioyijie, 2017-06-27:
@heimsi My CPU% is around 102% when i opened ZEDFU.