stereolabs/zed-ros-wrapper #135: Accessing Visual Odometry Algorithm

manomitbal, 2017-06-23:
Hi,

I am in the process of using the pose provided by the framework from the topic /zed/odom. I was wondering if it would be possible for me to modify (or set the pose of the camera) externally since I plan to implement an EKF to combine the pose estimates along with an IMU to prevent drift. I could not find any functionality to set or update the odometry information. Is there a way to do that?

Thanks

davidbsp, 2018-02-16:
On the same note, it would be really important to have the driver publishing the covariances on the nav_msgs/Odometry msg.

27Apoorva, 2018-02-22:
Hi @manomitbal  were you able to establish the covariance?
I am trying to use zed's visual odometry in EKF as well but the topic /zed/odom doesn't provide covariance.

matislovas, 2018-04-28:
Any updates on it?

WASCHMASCHINE, 2018-04-30:
I think the covariance are provided in a recent version, but Stereolabs will not disclose their VO implementation.

fabolhak, 2018-05-25:
Hey,
there is a PR which implements static covariances:
https://github.com/stereolabs/zed-ros-wrapper/pull/192/files

nathangeorge1, 2019-05-03:
@manomitbal @davidbsp @27Apoorva if you want covariances for robot_pose_ekf just add this:

```
nav_msgs::Odometry camera_odom_out = odom_camera_in_bus.getData();
	camera_odom_out.pose.covariance[0]  = 1e-6;
	camera_odom_out.pose.covariance[7]  = 1e-6;
	camera_odom_out.pose.covariance[14] = 1e-6;
	camera_odom_out.pose.covariance[21] = 1e-6;
	camera_odom_out.pose.covariance[28] = 1e-6;
	camera_odom_out.pose.covariance[35] = 1e-6;
	camera_odom_out.twist.covariance = camera_odom_out.pose.covariance;
odom_camera_out_bus.broadcast(camera_odom_out);
```

Although my navigation software supports ZED; so it might be easier to use it: https://www.newmindrobotics.com/

Myzhar, 2019-05-03:
Static covariance option has been added to the latest `master` version: see the parameters `fixed_covariance` and `fixed_cov_value` in `common.yaml`:
https://www.stereolabs.com/docs/ros/zed_node/#zed-parameters

nathangeorge1, 2019-05-03:
O thanks; didn't see that