stereolabs/zed-ros-wrapper #136: Problem accessing image /depth/depth_registered/compressed

heimsi, 2017-06-28:
As soon as I try to access the compressed depth image of the topic _/depth/depth_registered/compressed_, e.g. with rqt, I get the following error:

`[ERROR] [1498638280.471027032]: Compressed Image Transport - JPEG compression requires 8-bit, 1/3-channel images (input format is: 32FC1)`

adujardin, 2017-06-28:
Hi,
In this wrapper, the depth can be either in 32-bit float (typically in meters) or in 16-bit unsigned char (in mm), with [openni_depth_mode](https://github.com/stereolabs/zed-ros-wrapper/blob/master/launch/zed_camera.launch#L19). 

I'm not sure if the compressed image mechanism in ROS is made for float data but rather for color or gray images (which would explain why it requires an 8-bit 1/3C). In the case of the depth, I think it's for normalized displayable only image.

heimsi, 2017-06-28:
I just discovered in [ROS REP 118](http://www.ros.org/reps/rep-0118.html) that the 16-bit unsigned openni_depth_mode representation should be compressable with via PNG:

>  Using sensor_msgs/Image already permits reasonable compression of 16-bit depth images with PNG, and easily allows adding compression algorithms specialized for depth images. 
 
[Under _Why Not sensor_msgs/DisparityImage_]

I just discovered as well that there is even a dedicated plugin to image_transport: [compressed_depth_image_transport](http://wiki.ros.org/compressed_depth_image_transport)

closed 2018-06-28 by Myzhar