stereolabs/zed-ros-wrapper #144: [error] depth_stabilization

james-yoo, 2017-08-14:
Hi,

I've got following error while 'catkin_make'  on Jetson TX2(Jetpack 3.1) board:

/home/nvidia/catkin_ws/src/zed-ros-wrapper/src/zed_wrapper_nodelet.cpp:689:19: error: ‘class sl::InitParameters’ has no member named ‘depth_stabilization’
             param.depth_stabilization = depth_stabilization;

So I commented out line 689 and recompile zed-ros-wrapper, it works.
Does anyone having the same issue?

carriojazz, 2017-08-30:
Same here. I fixed it by commenting the line:
param.depth_stabilization = depth_stabilization;

adujardin, 2017-08-31:
Hi,
`depth_stabilization` was introduced in the [ZED SDK 2.1](https://www.stereolabs.com/developers/release/2.1/).
You are probably using the ZED SDK 2.0

closed 2017-09-05 by james-yoo