stereolabs/zed-ros-wrapper #145: zed_wrapper_node crashes (exit code -9)

janblumenkamp, 2017-08-15:
Hello everyone,
I have a ZED camera connected to a Jetson TX1 via the USB 3.0 port. I installed ROS kinetic, the ZED SDK and put the ROS wrapper into the catkin workspace.
After launching the wrapper, it crashes after a short time. It claims that the camera is not connected to a USB 3 Port, but it is:

```
lsusb -t 
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=tegra-xhci/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Vendor Specific Class, Driver=r8152, 5000M
    |__ Port 2: Dev 4, If 0, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 2: Dev 4, If 1, Class=Video, Driver=uvcvideo, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=tegra-xhci/5p, 480M
```
This is the log from the console beginning with the launch of the wrapper and ending with Ctrl-C after the process died.

```
roslaunch zed_wrapper zed.launch 
... logging to xxxx.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server 

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /zed/zed_description: <?xml version="1....
 * /zed/zed_wrapper_node/confidence: 100
 * /zed/zed_wrapper_node/depth_cam_info_topic: depth/camera_info
 * /zed/zed_wrapper_node/depth_stabilization: 1
 * /zed/zed_wrapper_node/depth_topic: depth/depth_regis...
 * /zed/zed_wrapper_node/frame_rate: 60
 * /zed/zed_wrapper_node/gpu_id: -1
 * /zed/zed_wrapper_node/left_cam_info_topic: left/camera_info
 * /zed/zed_wrapper_node/left_raw_topic: left/image_raw_color
 * /zed/zed_wrapper_node/left_topic: left/image_rect_c...
 * /zed/zed_wrapper_node/odometry_db: 
 * /zed/zed_wrapper_node/odometry_topic: odom
 * /zed/zed_wrapper_node/openni_depth_mode: 0
 * /zed/zed_wrapper_node/point_cloud_topic: point_cloud/cloud...
 * /zed/zed_wrapper_node/quality: 1
 * /zed/zed_wrapper_node/resolution: 2
 * /zed/zed_wrapper_node/rgb_cam_info_topic: rgb/camera_info
 * /zed/zed_wrapper_node/rgb_raw_topic: rgb/image_raw_color
 * /zed/zed_wrapper_node/rgb_topic: rgb/image_rect_color
 * /zed/zed_wrapper_node/right_cam_info_topic: right/camera_info
 * /zed/zed_wrapper_node/right_raw_topic: right/image_raw_c...
 * /zed/zed_wrapper_node/right_topic: right/image_rect_...
 * /zed/zed_wrapper_node/sensing_mode: 0
 * /zed/zed_wrapper_node/svo_filepath: 
 * /zed/zed_wrapper_node/zed_id: 0

NODES
  /zed/
    zed_base_link1 (tf2_ros/static_transform_publisher)
    zed_base_link3 (tf2_ros/static_transform_publisher)
    zed_base_link4 (tf2_ros/static_transform_publisher)
    zed_state_publisher (robot_state_publisher/state_publisher)
    zed_wrapper_node (zed_wrapper/zed_wrapper_node)

ROS_MASTER_URI=http://xxxxxx:11311

core service [/rosout] found
process[zed/zed_base_link1-1]: started with pid [11661]
process[zed/zed_base_link3-2]: started with pid [11662]
process[zed/zed_base_link4-3]: started with pid [11663]
process[zed/zed_wrapper_node-4]: started with pid [11670]
process[zed/zed_state_publisher-5]: started with pid [11672]
[ INFO] [1502810732.356716906]: Initializing nodelet with 4 worker threads.
ZED (Init) >> Depth mode: Performance
ZED (Init) >> Video mode: HD720@30
ZED (Init) >> WARNING: Corrupted frames detected. Check that your ZED is connected to a USB3.0 port
[ INFO] [1502810742.092697986]: Error code: Low USB Bandwidth
ZED (Init) >> Depth mode: Performance
ZED (Init) >> Video mode: HD720@30
ZED (Init) >> WARNING: Corrupted frames detected. Check that your ZED is connected to a USB3.0 port
[ INFO] [1502810752.431304181]: Error code: Low USB Bandwidth
ZED (Init) >> Depth mode: Performance
ZED (Init) >> Video mode: HD720@30
ZED (Init) >> WARNING: Corrupted frames detected. Check that your ZED is connected to a USB3.0 port
[ INFO] [1502810763.166680187]: Error code: Low USB Bandwidth
ZED (Init) >> Depth mode: Performance
ZED (Init) >> Video mode: HD720@30
ZED (Init) >> WARNING: Corrupted frames detected. Check that your ZED is connected to a USB3.0 port
[ INFO] [1502810773.903099650]: Error code: Low USB Bandwidth
ZED (Init) >> Depth mode: Performance
ZED (Init) >> Video mode: HD720@30
ZED (Init) >> WARNING: Corrupted frames detected. Check that your ZED is connected to a USB3.0 port
[ INFO] [1502810784.425901150]: Error code: Low USB Bandwidth
ZED (Init) >> Depth mode: Performance
ZED (Init) >> Video mode: HD720@30
ZED (Init) >> WARNING: Corrupted frames detected. Check that your ZED is connected to a USB3.0 port
[ INFO] [1502810795.299867442]: Error code: Low USB Bandwidth
ZED (Init) >> Depth mode: Performance
ZED (Init) >> Video mode: HD720@30
ZED (Init) >> WARNING: Corrupted frames detected. Check that your ZED is connected to a USB3.0 port
[ INFO] [1502810806.323154849]: Error code: Low USB Bandwidth
ZED (Init) >> Depth mode: Performance
ZED (Init) >> Video mode: HD720@30
ZED (Init) >> WARNING: Corrupted frames detected. Check that your ZED is connected to a USB3.0 port
[ INFO] [1502810817.190281523]: Error code: Low USB Bandwidth
ZED (Init) >> Depth mode: Performance
ZED (Init) >> Video mode: HD720@30
[zed/zed_wrapper_node-4] process has died [pid 11670, exit code -9, cmd /xxx/catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node __name:=zed_wrapper_node __log:=/xxx/.ros/log/daecbbac-81c4-11e7-ab29-bdaee697de70/zed-zed_wrapper_node-4.log].
log file: /xxx/.ros/log/daecbbac-81c4-11e7-ab29-bdaee697de70/zed-zed_wrapper_node-4*.log
^C[zed/zed_base_link4-3] killing on exit
[zed/zed_state_publisher-5] killing on exit
[zed/zed_base_link1-1] killing on exit
[zed/zed_base_link3-2] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```
Any idea what could be wrong?

janblumenkamp, 2017-10-11:
This problem also appears on a different setup (Jetson TX1 with Auvidea J140 carrier and differenz ZED camera). This is the log file from the original launch file, nothing changed; launched with roslaunch zed_wrapper zed.launch.
[roslaunch-aila2-head-jetson-1971.log](https://github.com/stereolabs/zed-ros-wrapper/files/1375710/roslaunch-aila2-head-jetson-1971.log)

chan-e91, 2017-10-20:
Hi, I have used same config(Tx1 + ZED, using ZED ROS wrapper), is the issue resolved ?

janblumenkamp, 2017-10-21:
I have written a message to the stereolabs support and got this response:

> First of all, I'll recommend you to test the release candidate of the upcoming version of the SDK that can be downloaded here (RC4): https://www.stereolabs.com/download_QA/
> 
> On the TX1, the USB 3.0 communication is limited in terms of bandwidth, making HD resolutions and high framerates unavailable.
> Nvidia solved this issue on the TX2.
> 
> We are recommending the VGA@60fps resolution for the TX1.

With the RC4 SDK, the USB crashes stopped for me.

closed 2017-10-21 by janblumenkamp

chan-e91, 2017-10-22:
Thanks for the information, :), I will try to use the latest version rc4
with vga @30fps and check the results.

https://www.stereolabs.com/download_QA/

MrDadaGuy, 2018-06-21:
I am having this same issue on CTI Rudi (TX2) and ZED.  I am using zed sdk 2.4.1