stereolabs/zed-ros-wrapper #146: Using gtx 1050ti with ZED SDK 1.1.0 on Jackal Robot.

hman2005, 2017-08-21:
I am trying to get the ZED working when attached to a Clearpath Jackal robot.  The robot has it's own local computer that runs on Ubuntu 14.04, so I was forced to use ZED SDK 1.1.0.  The robot is designed to run with ROS, so it already had ROS Indigo installed.  This led to some issues with OpenCV compatability, but I fixed that by dual installing OpenCV 3.1 alongside OpenCV 2.4.8 (the version that was there initially).  I will also mention that my cuda version is 7.5.  Lastly, we hooked a GeForce GTX 1050Ti up to the robot, as it didn't come with its own GPU.

In list form:
ZED SDK 1.1.0
ROS Indigo
OpenCV 3.1
Cuda 7.5
GTX 1050Ti

Now, whenever I try launching the ros wrapper, it is telling me that my there is no compatible GPU available.  It looks like this: 
```
core service [/rosout] found
process[zed_base_link1-1]: started with pid [7395]
process[camera/zed_wrapper_node-2]: started with pid [7396]
[ INFO] [1503338922.074579378]: ZED_WRAPPER Node initialized
[ INFO] [1503338922.637904332]: Using ZED Camera
ZED SDK >> (Init) Best GPU Found : GeForce GTX 1050 Ti , ID : 0
ZED SDK >> (Init) Disparity mode has been set to PERFORMANCE
ZED SDK >> (Init) Creating ZED GPU mem...
ZED SDK >> (Init) Starting Self-Calibration in background... 
ZED SDK >> (Init) No compatible NVIDIA device found, disparity map unavailable. 
 Err Code : 8 : 0
[ INFO] [1503338930.522934143]: NO_GPU_COMPATIBLE
ZED SDK >> (Init) Best GPU Found : GeForce GTX 1050 Ti , ID : 0
ZED SDK >> (Init) Disparity mode has been set to PERFORMANCE
 
ZED SDK >> (Init) Creating ZED GPU mem...
ZED SDK >> (Init) A lot of corrupted frames (frame glitch or invalid size) has been detected. Please check that your ZED is connected to an USB3.0 port, that the USB3.0 chipset driver is up to date...
ZED SDK >> (Init) Starting Self-Calibration in background... 
ZED SDK >> (Init) No compatible NVIDIA device found, disparity map unavailable. 
 Err Code : 8 : 0
[ INFO] [1503338944.381466652]: NO_GPU_COMPATIBLE
```
Running ZED Diagnostic gives me the following: 

```
CUDA Capabilities

 CUDA Check
(SMI) : Driver Version                      : 375.20
(SMI) :     Driver Model
(SMI) :     Performance State               : P0
cuDeviceGetCount : OK : found 1 NVIDIA Graphic card. Taking dev 0
cuDeviceGetName : OK : GeForce GTX 1050 Ti
cuDeviceComputeCapability :  Compute Cap :  6 1 -- must be between 2.0 and 5.3
cuDriverGetVersion :  CUDA Driver version : 8000 . Must be >=7500 for CUDA 7.5 toolkit.
cuDeviceTotalMem:   CUDA Memory  4069 MBytes (4266590208 bytes)
```
[This guy](https://github.com/stereolabs/zed-ros-wrapper/issues/84) had a pretty similar issue, and they said that he has to either upgrade Ubuntu to 16.04 and CUDA to 8.0, or to get a new, older GTX.  Are these really my only options?  I can't upgrade without potentially breaking everything, so I would have to buy a new GPU.

Also, I think its worth mentioning that, as the robot doesn't have a screen, I can't run most of the built-in tools.

Myzhar, 2018-06-28:
@hman2005 have you solved this issue?
As far as I know some Ubuntu systems do not start graphical environment if there is not a screen connected to HDMI port, this is why you get "No compatible NVIDIA device found". 

There are a lot of ways to solve the problem, let me know if you need a solution anymore.

Myzhar, 2018-09-08:
I close the issue but you can continue to post comments if you need further help

closed 2018-09-08 by Myzhar