stereolabs/zed-ros-wrapper #147: Stereo images combined

pavloblindnology, 2017-08-22:
Hello,

ZED ROS wrapper publishes stereo images individually.
When you start to record them using rosbag, issues may arise.
For example, sometimes, one of the images from the stereo pair can be dropped by rosbag.
Or sometimes image can be dropped when playing back a rosbag.
Such issues are easily overcome by publishing a combined stereo pair - one double-sized image with left and right images put side by side (horizontally or vertically).
In this case you also don't need to use ROS message filtering to sync image pairs when subscribing.
Can you please add such feature in the future release?

Myzhar, 2018-07-12:
We are not planning to add this feature for the moment, but you can get the to rectified images combined side-by-side using the value `VIEW_SIDE_BY_SIDE` (https://www.stereolabs.com/developers/documentation/API/v2.4.0/group__Video__group.html#ga77fc7bfc159040a1e2ffb074a8ad248c) with the `retrieveImage` function:
https://www.stereolabs.com/developers/documentation/API/v2.4.0/classsl_1_1Camera.html#ac40f337ccc76cacd3412b93f7f4638e2

WASCHMASCHINE, 2018-07-13:
Is it actually faster to retrieve it SIDE_BY_SIDE and cropping it to cv::Mat than retrieving and converting both?

Myzhar, 2018-07-13:
I think that it takes a similar time. Let us know if you make a test.

pavloblindnology, 2018-07-13:
What about retrieving grayscale side_by_side from ZED?
Also, it'd be nice to have zed_ros_wrapper being able to publish grayscale side_by_side to avoid ROS synchronization issues.

WASCHMASCHINE, 2018-07-13:
I've tested it and it makes only a small difference in microseconds if you have to use SIDE_BY_SIDE (slightly slower), I assume mostly because of the extra step from 1 to 2 cv::Mats

closed 2018-11-24 by Myzhar

pavloblindnology, 2018-11-26:
@WASCHMASCHINE It's not about speed, it's about having consistent stereo pairs and dropping the frames.

WASCHMASCHINE, 2018-11-26:
@pavloblindnology SIDE_BY_SIDE should be always consistent. If you are not fast enough, then you are dropping frames, I assume.

pavloblindnology, 2018-11-26:
@WASCHMASCHINE That is the point, SIDE_BY_SIDE is consistent. That is why I asked about adding it to ROS wrapper. And having SIDE_BY_SIDE in grayscale is even better for stereo purposes.

WASCHMASCHINE, 2018-11-27:
@pavloblindnology I agree that returning left and right images separately via a rostopic is a bad idea for that reason. I think you can easily modify the code, which is what I did, too. So, I do not understand what you are trying to accomplish here.

Stereolabs might not be genereally interested in supporting non-SDK stereo. I also think they fear that if they disclose the algorithms that they degrade to only another "hardware supplier", but now they can also sell you the software.

pavloblindnology, 2018-11-27:
@WASCHMASCHINE And I did it. But it'd be much better to have such easily added and needed features to be actually added to official ZED ROS wrapper, to save your time on smth more meaningful. So, it's an ordinary feature request.