stereolabs/zed-ros-wrapper #152: How to change video mode

shank1608, 2017-09-16:
When I run zed launch  on ros it takes 1080 @2 video mode , I want to get 720 at higher frame rate. How can I get that?

cosama, 2018-01-22:
As I understand from [Getting Started](https://www.stereolabs.com/documentation/integrations/ros/getting-started.html) and [ZED node](https://www.stereolabs.com/documentation/integrations/ros/ZED_node.html) it should be:

`roslaunch zed_wrapper zed.launch resolution:=2 frame_rate=60`

However, it seems to not work. It also is not possible to reconfigure these parameters later with
 `rosrun dynamic_reconfigure dynparam set /zed/zed_wrapper_node resolution 2`

The only way to change stuff is to edit the *.launch files, but because they call each other you have to edit multiple files in certain situation.

I would really appreciate, if these settings could simply be changed on the command line. I have issues with frame rates in ros rviz and I can not find an easy way of trying different resolutions/frame rates/algorithm easily for debugging it.

(ZED SDK 2.3, ros kinetics, ubuntu)

Myzhar, 2018-06-28:
Fixed in https://github.com/stereolabs/zed-ros-wrapper/commit/4870071e89903e25408409e4d9e116cd4f22704a

closed 2018-06-28 by Myzhar