stereolabs/zed-ros-wrapper #157: OpenCV core not found

jihobak, 2017-10-02:
My env is below
```
-tk1
-Ubuntu 14.04
-indigo(ros)
```
when I follow this site, http://wiki.ros.org/zed-ros-wrapper
```
cd ~/catkin_ws/
catkin_make
```

I got an error.
```
CMake Error at zed-ros-wrapper/CMakeLists.txt:17 (message):
  
 OpenCV core not found, install it from the tutorial at:
 https://www.stereolabs.com/documentation/overview/getting-started/getting-started.html 

Call Stack (most recent call first):
  zed-ros-wrapper/CMakeLists.txt:37 (checkPackage)
```

I already installed opencv4tegra from jetpack...

Is there anything I should check for using opencv?

Thanks

adujardin, 2017-10-02:
Hi,
Which version of zed-ros-wrapper did you use? 
For the TK1, the [ZED SDK 1.2](https://www.stereolabs.com/developers/downloads/archives/ZED_SDK_Linux_JTK1_v1.2.0.run) is required alongside [zed-ros-wrapper for 1.2](https://github.com/stereolabs/zed-ros-wrapper/releases/tag/v1.2.0)

jihobak, 2017-10-03:
@adujardin I got this. Thanks for you help :)

closed 2017-10-03 by jihobak