stereolabs/zed-ros-wrapper #158: Question about zed node(zed node often die)

jihobak, 2017-10-03:
After basic install,
```
roslaunch zed_wrapper zed.launch
```
It works but the problems is it works almost randomly.
I tried reconnect usb and zed camera... I still don't know what the problems is.
error message is below.

In addition, 'ZED SDK >> (Init) Starting Self-Calibration in background... ' this stage take almost ~ 5min when It work. Is it normal?
```
started core service [/rosout]
process[zed/zed_base_link1-2]: started with pid [11480]
process[zed/zed_base_link3-3]: started with pid [11484]
process[zed/zed_wrapper_node-4]: started with pid [11494]
process[zed/zed_state_publisher-5]: started with pid [11503]
[ INFO] [1506999484.817437770]: Initializing nodelet with 4 worker threads.
[ INFO] [1506999485.921131903]: Using ZED Camera
ZED SDK >> (Init) Best GPU Found : GK20A , ID : 0
ZED SDK >> (Init) Disparity mode has been set to PERFORMANCE
ZED SDK >> (Init) Creating ZED GPU mem...
ZED SDK >> (Init) Starting Self-Calibration in background... 
[zed/zed_wrapper_node-4] process has died [pid 11494, exit code -9, cmd /home/ubuntu/catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node __name:=zed_wrapper_node __log:=/home/ubuntu/.ros/log/aca66e62-a7e6-11e7-a2a7-00044b709381/zed-zed_wrapper_node-4.log].
log file: /home/ubuntu/.ros/log/aca66e62-a7e6-11e7-a2a7-00044b709381/zed-zed_wrapper_node-4*.log

```

my env:
-tk1
-ubuntu 14
-ros version: indigo
-zed sdk 1.2.0

my ZED_Diagonostic_log.txt
```

*******************************
*** OS Version ***
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.5 LTS
Release:	14.04
Codename:	trusty

Linux tegra-ubuntu 3.10.40-ga7da876 #1 SMP PREEMPT Tue Jun 7 21:05:06 PDT 2016 armv7l armv7l armv7l GNU/Linux

*******************************
*** Hardware ***

*******************************
*** USB Connection List ***
>> Detected USB3.0 component : 1
>> Detected ZED with PID/VID : 1
>> USB : Bus 002 Device 011: ID 2b03:f580  
Bus 002 Device 008: ID 2109:0812 VIA Labs, Inc. VL812 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 028: ID 046d:c05a Logitech, Inc. M90/M100 Optical Mouse
Bus 001 Device 027: ID 045e:0752 Microsoft Corp. Wired Keyboard 400
Bus 001 Device 026: ID 2109:2812 VIA Labs, Inc. VL812 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

*******************************

*******************************
*** ZED Capabilities***

ZED Has been detected with PID:VID : 
bLength : 18
bDescriptorType : 1
bcdUSB : 768
bDeviceClass : 239
bDeviceSubClass : 2
bDeviceProtocol : 1
bMaxPacketSize0 : 9
idVendor : 11011
bcdDevice : 256
iManufacturer : 1
iProduct : 2
iSerialNumber : 0
bNumConfigurations : 1

 New Video Capture device detected : 
capability.driver : uvcvideo
capability.card : ZED
fmtdesc.description: YUV 4:2:2 (YUYV)

 OK : ZED is read under /dev/video0
>> setupDevice for 672x376@15 result : 1  (Note : returned Size was: 672x376)
>> setupDevice for 672x376@30 result : 1  (Note : returned Size was: 672x376)
>> setupDevice for 672x376@60 result : 1  (Note : returned Size was: 672x376)
>> setupDevice for 672x376@100 result : 1  (Note : returned Size was: 672x376)
>> setupDevice for 1280x720@15 result : 1  (Note : returned Size was: 1280x720)
>> setupDevice for 1280x720@30 result : 1  (Note : returned Size was: 1280x720)
>> setupDevice for 1280x720@60 result : 1  (Note : returned Size was: 1280x720)
>> setupDevice for 1920x1080@15 result : 1  (Note : returned Size was: 1920x1080)
>> setupDevice for 1920x1080@30 result : 1  (Note : returned Size was: 1920x1080)
>> setupDevice for 2208x1242@15 result : 1  (Note : returned Size was: 2208x1242)

 OK : ZED can setup all resolutions

 ZED Serial Number : 14620
 ZED Firmware Number : 1142

*******************************
*** CUDA Capabilities***

 *** CUDA Check *** 
cuDeviceGetCount : OK : found 1 NVIDIA Graphic card. Taking dev 0
cuDeviceGetName : OK : GK20A
cuDeviceComputeCapability :  Compute Cap :  3 2 -- must be 3.2 for Jetson TK1 
cuDriverGetVersion :  CUDA Driver version : 6050 . Must be >=6500 for CUDA 6.5 toolkit (Jetson TK1).
cuDeviceTotalMem:   CUDA Memory  1892 MBytes (1984385024 bytes)
```

Thanks a lot.

closed 2018-06-28 by Myzhar

jgegit, 2018-07-18:
Did you solve this issue? I have the same problem. Same diagnostic log.

Myzhar, 2018-07-19:
Are you using the latest SDK and wrapper versions?