stereolabs/zed-ros-wrapper #159: Cannot launch ZED when I boot Ubuntu without a screen

ChrisPsa, 2017-10-16:
Roslaunch results in `Error code:  Camera Not Detected` when booting in Ubuntu without a screen. I noticed there is a no-screen branch. I don't know if its purpose is to make it work without a screen, but it is a very old branch and it does not compile with the latest ZED SDK.

adujardin, 2017-10-16:
Hi,
It's a known bug. The upcoming release of the ZED SDK (2.2) solves this issue. 
The `no-screen` branch is obsolete now and will be removed.

ChrisPsa, 2017-10-16:
Thank you for the speedy response. When is ZED SDK (2.2) expected to be released?

adujardin, 2017-10-16:
It should be out in a couple of weeks, it's currently in QA.
You can get in touch with the support (at support@stereolabs.com) to get a beta.

closed 2017-10-16 by ChrisPsa