stereolabs/zed-ros-wrapper #160: How to use SDK with the ZED camera connected in another computer via wifi?

whstarxu54, 2017-10-20:
In my application, the ZED camera is connected on one computer with no GPUs while my program is running on another more powerful computer with GPUs that are able to use the ZED SDK. Communication between the two computers are through wifi. 
I would like to first receive the video data from ZED via wifi and then use ZED SDK to do further processing. How can I achieve this?

adujardin, 2017-10-20:
Hi,
Unfortunately, this is not possible. The ZED SDK has to run on the same PC as the connected ZED.

closed 2017-10-24 by nesnes

taraprasad73, 2019-03-12:
Can we not do this? Capture images through UVC on the non-GPU system, send the images through wifi and process the images in the system having GPU with ZED SDK.

Myzhar, 2019-03-12:
@taraprasad73 the ZED SDK does not accepts images as real time input. 
Only SVO files can be used to perform post processing not in real time:
https://www.stereolabs.com/docs/video/#video-recording