stereolabs/zed-ros-wrapper #163: ./ZED Depth Viewer: error while loading shared libraries: libnppi.so.7.5: cannot open shared object file: No such file or directory

puren, 2017-11-07:
Hello,

It is not directly related to ros wrapper. But I want to use that too, eventually. So, it can be implicitly related. 
I installed ZED stereo camera SDK 2.2 for Cuda 9.0 in Ubuntu 16.04.
./ZED Explorer works without any problem. However, when I run ./'ZED Depth Viewer', it gives the following error:
`./ZED Depth Viewer: error while loading shared libraries: libnppi.so.7.5: cannot open shared object file: No such file or directory`
libnppi.so.7.5 is not a part of Cuda 9.0 and as the ZED SDK that I installed is for Cuda 9.0. So, I don't understand why it looks for that library.
Could you give me any suggestion to resolve this issue?

adujardin, 2017-11-07:
Hi,
Yes, an issue was found with the Ubuntu installer for ZED SDK 2.2 Cuda 9. We are working on it. In the meantime, if possible you should use the Cuda 8 version which doesn't have this bug.

puren, 2017-11-07:
Thank you for your answer. I'll do that.

closed 2017-11-07 by puren