stereolabs/zed-ros-wrapper #164: Problems about two zed stereo cameras calibration

bellqlz, 2017-11-10:
Hi,

I want to do mapping by two zed cameras so that I used Opencv cameracalibrate to get cameramatrix and distortion coefficients of each camera and stereocalibrate to acquire rotation matrix and transformation matrix which I used to launch two cameras together to do mapping. However the mapping results are bad -- there are still two objects. The error about cameracalibrate and stereocalibrate are around 0.5 and tranformation matrix is correct that I made two cameras parrallel and their distance is same as in the transformation matrix. Did I do wrong steps or zed cannot do multicamera calibration?

WASCHMASCHINE, 2017-11-10:
Hi, 
I do not understand your results and your setup.

You have two ZED cameras, so a total of four cameras. You did OpenCV's stereo calibration on the left camera of both ZEDs. Then you did what? You "mapped" one object and now there are two? 
What does the error mean?

bellqlz, 2017-11-13:
Hi,
When I completed stereocalibration, I obtained the transformation matrix and rotation matrix between two stereo cameras. And then, I applied these two matrix on multicamera mapping to align two stereo cameras in order to get the results of two stereo cameras can match enough that there was only one object which seems mapped by a single stereo camera. However, the result I got was bad that they didn't match and there were still two objects.

WASCHMASCHINE, 2017-11-13:
Shouldn't you need only one transformation matrix (rotation + translation) from camera 2 to camera 1? Then you use the point cloud? Have you calibrated both cameras? How bad was the result and what did you expect?

closed 2018-06-28 by Myzhar