stereolabs/zed-ros-wrapper #166: Error code: Camera Not Detected

lhs55349780, 2017-11-20:
Hi,

I'm using ZED camera on ubuntu 16.04 LTS, nvidia TX2 board. 
Originally, roslaunch ZED_wrapper zed.launch does work, but recently it does not work with this error message:
[ INFO] [1511177185.346457156]: Error code: Camera Not Detected

Through lsusb command, I found that there's no ZED camera recognized in my computer.
What can I do to detect ZED Camera?

WASCHMASCHINE, 2017-11-20:
If the normal samples don't run, you should just contact the support from the website directly. This is not software related.

wujiang, 2017-12-20:
Sounds like it's hardware connection issue. If `lsusb` doesn't return the device, it means it's either not connected or the usb controller is not enabled. Can you post your `dmesg -Tw` output when you plug or unplug your ZED camera?

ChrisPsa, 2018-01-04:
The pre 2.2 ZED SDK had this issue on Ubuntu 16.04 when the OS was started without a screen attached (apparently it had something to do with the Xserver?). Anyway, updating to latest ZED SDK should fix the problem.

chan-e91, 2018-02-21:
I have got same error , when ZED was connected Ubuntu 16.04, it was working fine on Jetson TX1, but when connected to laptop, it throws following error "ZED camera not found".
The output of "dmesg -Tw" is 
"[Thu Feb 22 04:36:51 2018] usb 2-1: USB disconnect, device number 3
[Thu Feb 22 04:36:57 2018] usb 1-3: new high-speed USB device number 5 using xhci_hcd
[Thu Feb 22 04:36:57 2018] usb 1-3: New USB device found, idVendor=2b03, idProduct=f580
[Thu Feb 22 04:36:57 2018] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[Thu Feb 22 04:36:57 2018] usb 1-3: Product: ZED
[Thu Feb 22 04:36:57 2018] usb 1-3: Manufacturer: Leopard
[Thu Feb 22 04:36:57 2018] uvcvideo: Found UVC 1.00 device ZED (2b03:f580)
[Thu Feb 22 04:36:57 2018] uvcvideo 1-3:1.0: Entity type for entity ZED was not initialized!
[Thu Feb 22 04:36:57 2018] uvcvideo 1-3:1.0: Entity type for entity ZED was not initialized!
[Thu Feb 22 04:36:57 2018] uvcvideo 1-3:1.0: Entity type for entity Camera 1 was not initialized!
"

ncnynl, 2018-06-26:
I  aslo  have this issue .   

dmsg -TW  

For good one:
[Fri Feb 12 01:43:27 2016] uvcvideo: Found UVC 1.00 device ZED (2b03:f580)
[Fri Feb 12 01:57:02 2016] usb 1-2.4: new low-speed USB device number 13 using xhci-tegra
[Fri Feb 12 01:57:02 2016] usb 1-2.4: New USB device found, idVendor=093a, idProduct=2510
[Fri Feb 12 01:57:02 2016] usb 1-2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[Fri Feb 12 01:57:02 2016] usb 1-2.4: Product: USB Optical Mouse
[Fri Feb 12 01:57:02 2016] usb 1-2.4: Manufacturer: PixArt
[Fri Feb 12 01:57:03 2016] usb 1-2.4: ep 0x81 - rounding interval to 64 microframes, ep desc says 80 microframes
[Fri Feb 12 01:57:03 2016] input: PixArt USB Optical Mouse as /devices/3530000.xhci/usb1/1-2/1-2.4/1-2.4:1.0/0003:093A:2510.0004/input/input14

For bad One:

[Fri Feb 12 01:14:20 2016] uvcvideo: Found UVC 1.10 device ZED (2b03:f582)
[Fri Feb 12 01:14:20 2016] input: ZED as /devices/3530000.xhci/usb2/2-1/2-1.4/2-1.4:1.0/input/input13
[Fri Feb 12 01:31:25 2016] pwm_fan_driver pwm-fan:  Enabled vdd-fan
[Fri Feb 12 01:43:12 2016] usb 1-2.1.3: USB disconnect, device number 12
[Fri Feb 12 01:43:15 2016] usb 2-1.4: USB disconnect, device number 7
[Fri Feb 12 01:43:15 2016] xhci-tegra 3530000.xhci: tegra_xhci_mbox_work mailbox command 6
[Fri Feb 12 01:43:27 2016] usb 2-1.1.4: new SuperSpeed USB device number 8 using xhci-tegra
[Fri Feb 12 01:43:27 2016] usb 2-1.1.4: New USB device found, idVendor=2b03, idProduct=f580
[Fri Feb 12 01:43:27 2016] usb 2-1.1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[Fri Feb 12 01:43:27 2016] usb 2-1.1.4: Product: ZED
[Fri Feb 12 01:43:27 2016] usb 2-1.1.4: Manufacturer: Leopard
[Fri Feb 12 01:43:27 2016] xhci-tegra 3530000.xhci: tegra_xhci_mbox_work mailbox command 6

I use TX2  

 uname -a
Linux racebot 4.4.15 #2 SMP PREEMPT Thu Feb 1 10:35:03 CST 2018 aarch64 aarch64 aarch64 GNU/Linux

wujiang, 2018-06-26:
@ncnynl sometimes I experience the same issue. ZED drops connection sporadically.

Myzhar, 2018-06-28:
@ncnynl @wjzhou Can you please add information about what combination of FW, SDK, device and Operating System cause this kind of issue?

ncnynl, 2018-07-03:
@Myzhar    

ZED newest  FW  is v1523 
lsusb  information is:  uvcvideo: Found UVC 1.10 device ZED (2b03:f582)

ZED last old FW is v1142
lsusb  information is:  Found UVC 1.00 device ZED (2b03:f580)

On TX2 with FW v1523  
run roslaunch ZED_wrapper zed.launch  can not detect ZED Camera  

So need downgrade FW from v1523 to  v1142 

How to downgrade 
install Newest ZED SDK ， open ZED Explorer ， choose  firmware  v1142 
SEE：https://github.com/stereolabs/zed-examples/issues/63

Myzhar, 2018-07-03:
You can find the correct configuration for FW version and SDK version following this link:
https://support.stereolabs.com/hc/en-us/articles/360005189714-How-do-I-update-my-ZED-camera-firmware-

plieningerweb, 2018-08-14:
Porblem was solved for me after message "CAMERA DETECTION ISSUE" by flipping the usb -C connector 180degree and plugging it in again.

VGrondin, 2019-10-18:
I had this error when launching ZEDfu in live mode :
`ZED >> ERROR: sl::Camera::Open has not been called, no Camera instance running.`
Solved it by using another USB 3.0 connector. The one in front wasn't working but when i plugged it at the back of the tower it worked !

zoombinis, 2020-12-09:
I am also getting this issue where Zed only seems to work on Ubuntu 18.04 (docker container) if an X11 Client is connecting to the X11 Server on my host machine.

I run Zed Explorer but no updates to firmware appear available. I currently have version `1523`

I've tried plugging usb in and out but this really seems to be an issue somehow related to X11. For instance, I get this error only when running demos that have no visual gui like:

https://www.stereolabs.com/docs/ros/zed-nodelets/#examples

Myzhar, 2020-12-09:
@zoombinis the ZED camera requires CUDA.
CUDA must be correctly configured to run in a docker container, otherwise the ZED SDK cannot work.
The version of the FW is not your issue.
Please read more about how to properly use Docker with ZED: https://www.stereolabs.com/docs/installation/docker/

zoombinis, 2020-12-09:
@Myzhar my comment above lists I'm using cuda 10.2 and that the Zed SDK is fully operational.

My problem is I can *only get the Zed working if the docker container it runs in is connected to the X11 server of my host machine*

So I do not think this is cuda related, or if it is can you please clarify why you think that?

zoombinis, 2020-12-09:
@Myzhar not cuda, but you're right that I was missing --privileged mode and I thought that was only necessary for trying to run X11, but no it's necessary for any container running zed sdk.

Thanks!

Myzhar, 2020-12-09:
I'm glad that you solved your problem.
`--privileged` is indeed required to correctly use CUDA with the ZED SDK inside a container.