stereolabs/zed-ros-wrapper #167: How can I switch off an automatic exposure setting?

lastflowers, 2017-11-28:
I am using ZED in Ubuntu 14.04 (ros indigo) with cuda 7.5 and ZED SDK 1.0.0 and zed-ros-wrapper 1.0.0

To prevent sudden changes in intensity, I'd like to switch off automatic exposure setting for the camera.

I found that in ZED Explorer there is a "auto" button to switch off in "Settings" tab.
But, If I switch off the "auto" in the ZED Explorer and then run the wrapper, I've got too bright image streams even if I didn't adjust other settings.

Is it possible to turn off automatic exposure setting using zed.launch file ? or other way ?

Thanks in advance.

adujardin, 2017-11-28:
Hi,
It is possible to turn it off or set a specific value but not out of the box. You will need to add a function that calls : 

- [Camera::setCameraSettingsValue](https://www.stereolabs.com/developers/documentation/API/v1.0.0/classsl_1_1zed_1_1Camera.html#ada9ea1ff360615ef0f41fc4f4816094c) : `setCameraSettingsValue(ZEDCamera_settings::ZED_EXPOSURE, 0)`
- you can also [set a specific value](https://www.stereolabs.com/developers/documentation/API/v1.0.0/group__Enumerations.html#ga4558a79b2aed89eb945a032b5fa7a1d7)

lastflowers, 2017-11-28:
Thanks a lot ^^

closed 2018-06-28 by Myzhar