stereolabs/zed-ros-wrapper #169: Raw Images Missing Distortion Parameters

wngreene, 2017-12-04:
There appears to be a bug in zed_wrapper_nodelet.cpp such that the raw image streams are published using the rectified distortion parameters (which are zeros since the rectified streams are undistorted). You can see here where `publishCamInfo` is called on line ~522:
```c++
                    // Publish the left_raw == rgb_raw image if someone has subscribed to
                    if (left_raw_SubNumber > 0 || rgb_raw_SubNumber > 0) {
                        // Retrieve RGBA Left image
                        zed->retrieveImage(leftZEDMat, sl::VIEW_LEFT_UNRECTIFIED);
                        cv::cvtColor(toCVMat(leftZEDMat), leftImRGB, CV_RGBA2RGB);
                        if (left_raw_SubNumber > 0) {
                            publishCamInfo(left_cam_info_msg, pub_left_cam_info, t);
                            publishImage(leftImRGB, pub_raw_left, left_frame_id, t);
                        }
                        if (rgb_raw_SubNumber > 0) {
                            publishCamInfo(rgb_cam_info_msg, pub_rgb_cam_info, t);
                            publishImage(leftImRGB, pub_raw_rgb, rgb_frame_id, t);
                        }
}
```
`left_cam_info_msg` is the same message object that is used in the rectified streams and does not include the distortion parameters. `fillCamInfo` only produces correct cam infos for the rectified streams.

adujardin, 2017-12-05:
Hi
Thanks for the report! Indeed, you're right. 
I will push a fix soon.

adujardin, 2017-12-05:
It should work now. Please let me know if you encounter an issue with this. Thanks !

commit e433d76a (2017-12-05): Fix #169 Publishing Raw parameters
closed 2017-12-05 by adujardin