stereolabs/zed-ros-wrapper #170: Are ZED TFs following the ROS standards?

plnegre, 2017-12-15:
Launching zed.launch and opening the rviz one can see this:

[image]

ROS official documentation says that:
[image]

In you code, you are changing the orientation of the pointcloud here: https://github.com/stereolabs/zed-ros-wrapper/blob/e433d76a7bc5d15b3cb35a84392ff6b90e2a746d/src/zed_wrapper_nodelet.cpp#L346

Is there any reason for this change?

adujardin, 2017-12-19:
Hi,

Yes, this part can be ambiguous and need an update. The standard followed is from http://www.ros.org/reps/rep-0103.html (for a body; x forward, y left, z up). 
We should add a second frame with the '_optical' suffix that follows the convention you described.

plnegre, 2017-12-20:
Thank you very much for the fast answer!

From my point of view:
1. The frame_id in the headers of the the left/right image topics as well as depth and points2 must have the coordinate frame: X right, Y down, Z out.

2. I think the axis orientation specification of the report you have cited is for rigid bodies in movement, like the base link of a robot. An example: imagine you have a turtlebot with 3 ZEDs: one looking forward, one looking left and one looking right. With the current frames of the ZED wrapper only the forward-looking camera accomplishes the specifications of [REP 103](http://www.ros.org/reps/rep-0103.html#axis-orientation), the other two do not. It is the responsibility of the robot to provide the transform between its sensors and the robot base_link.

A possible solution (just a suggestion, of course):
- Do not add more frames in your urdf, I think it would be very complicated for a camera.
- I would use only 3 frames:
   - left_optical: centered at the left camera optical center and with the correct orientation: X right, Y down, Z out.
   - right_optical: centered at the right camera optical center and with the correct orientation: X right, Y down, Z out.
   - zed_center: at the ZED housing center, with the orientation that you like best (my suggestion is use the same than the opticals, for the reason I exposed in point 2). 

With this solution, the frame_id of the left image topics, depth and points2 should be: left_optical. The frame_id of the right image topics should be right_optical. And all other topics related to camera odometry or tracking should have the frame_id: zed_center.

adujardin, 2017-12-20:
Thanks for your feedback. It is indeed something we overlook that need a refactor. 
I'll work on a prototype that includes your suggestions.

adujardin, 2018-01-24:
Hello again!

Following your suggestions, I pushed a prototype (on a fork [zed-ros-wrapper/](https://github.com/adujardin/zed-ros-wrapper/)). 
Feedbacks or pull requests are welcomed :)

@rbonghi since you updated the current version quite a lot in #143, feel free to leave any comments you might have about this change. 

(It might break the pipeline for quite a lot of users, so for now, it won't be pushed to the main repo branch.)

rajmohan747, 2018-05-24:
@plnegre  @adujardin   I am also facing the same issue at some point of robot calibration.
Currently X axis is giving the depth values.

For keeping the depth as Z axis is it ok to do like the following?

```
                point_cloud.points[i].z = cpu_cloud[i][2];
                point_cloud.points[i].y = -cpu_cloud[i][0];
                point_cloud.points[i].x = -cpu_cloud[i][1];
```

Myzhar, 2018-06-29:
ZED Tree has be reorganized according to the following structure (valid for ZED Mini, in ZED the imu_link is not present)
[image]
 
The new structure is available since commit 5d0d7fe1fda347d49f54faf23001b5376b1e8f19

closed 2018-07-02 by Myzhar

tianshiz, 2019-09-27:
sorry, a little confused. I noticed that point_cloud/cloud_registered topic is following the X-out convention instead of the Z-out axis convention for camera. Is this the only topic that is using the zed_left_camera_frame whereas all the other topics use zed_left(right)_camera_optical_frame?

I'm using display_zed.launch

Myzhar, 2019-09-27:
Hi @tianshiz 
The point cloud follows the ROS standard described in REP105:
https://www.ros.org/reps/rep-0105.html
The conversion from X-forward to Z- forward is performed by the SDK while creating the point cloud so no other calculation is required. 

PS do not reply to a closed issue, please open a new one even if your question seems related to it