stereolabs/zed-ros-wrapper #171: Depth image only in certain regions

ashnarayan13, 2017-12-15:
Hi, 

I wanted to know if the depth image can be computed only for certain regions of the image. 
I was looking through the code but couldn't find a clear method where I can tune the depth image result. 

Thanks!

WASCHMASCHINE, 2017-12-15:
Hi @ashnarayan13,

why would you want to do that? Improve performance? Just crop regions you need or ignore the rest.
Depending on the implementation, the algorithm may also not work that good locally.

ashnarayan13, 2017-12-16:
@WASCHMASCHINE 
[image]
I get the bounding box like this. I want to compute the depth only in that region and skip everything else. 
I am not sure which method in the zed wrapper handles this so I can add a region. 

Primarily, this is to reduce the area for computing depth which will save time and increase performance. 

Thanks!

WASCHMASCHINE, 2017-12-16:
Hi @ashnarayan13,
are you running in an actual bottleneck or is this premature optimization?

As far as I know, there is no option to limit the depth computation to a region.
Also, you would first have to send to the ROS node that you want to change the region again, but the car in the mean time did move a bit and the next frame was processed with the old region. So your region may now be off and you are missing a lot of pixels from the car.

My proposal would be to you to do all the image processing on the GPU directly on the ZED images and to not use ROS in your use case. You can tradeoff quality for more speed in the ZED API and you can also lower the resolution to get more FPS, as you might know.

ashnarayan13, 2017-12-18:
When I tried it on a laptop, I was getting around 10 fps with a mobile GPU on ROS. I got 30fps on the ZED API.

Since my current pipeline is built on ROS, I am not sure how I can do the depth processing on the API without ROS and then pass the depth to ROS. That should theoretically be faster as the ZED API is much faster than the ROS ZED wrapper.

closed 2018-06-28 by Myzhar