stereolabs/zed-ros-wrapper #175: CUDA_nppi_LIBRARY variables set to NOTFOUND

acrosson, 2018-01-10:
I recently upgraded cuda to 9.0 to use the latest Zed SDK and ROS package. Installation of the Zed package works correctly, however when I try to `catkin_make` I get the follow error:

```
-- +++ processing catkin package: 'zed_wrapper'
-- ==> add_subdirectory(zed_ros)
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
--   chrono
--   atomic
--   regex
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- looking for PCL_COMMON
-- looking for PCL_KDTREE
-- looking for PCL_OCTREE
-- looking for PCL_SEARCH
-- looking for PCL_IO
-- looking for PCL_SAMPLE_CONSENSUS
-- looking for PCL_FILTERS
-- looking for PCL_GEOMETRY
-- looking for PCL_FEATURES
-- looking for PCL_SEGMENTATION
-- looking for PCL_SURFACE
-- looking for PCL_REGISTRATION
-- looking for PCL_RECOGNITION
-- looking for PCL_KEYPOINTS
-- looking for PCL_VISUALIZATION
-- looking for PCL_PEOPLE
-- looking for PCL_OUTOFCORE
-- looking for PCL_TRACKING
-- looking for PCL_APPS
-- Could NOT find PCL_APPS (missing:  PCL_APPS_LIBRARY) 
-- looking for PCL_MODELER
-- looking for PCL_IN_HAND_SCANNER
-- looking for PCL_POINT_CLOUD_EDITOR
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDA_nppi_LIBRARY (ADVANCED)
    linked by target "zed_wrapper_node" in directory /home/user/project/src/zed_ros
    linked by target "zed_wrapper_node" in directory /home/user/project/src/zed_ros
    linked by target "ZEDWrapper" in directory /home/user/project/src/zed_ros
    linked by target "ZEDWrapper" in directory /home/user/project/src/zed_ros

-- Configuring incomplete, errors occurred!
See also "/home/user/project/build/CMakeFiles/CMakeOutput.log".
See also "/home/user/project/build/CMakeFiles/CMakeError.log".
```

Any suggestions?

acrosson, 2018-01-11:
I ended up downgrading to my previous kernel `4.10.0-42-generic`. and then installed NVIDIA driver `NVIDIA-Linux-x86_64-384.90.run` , CUDA 8.0 and the Zed SDK `ZED_SDK_Linux_Ubuntu16_v2.1.2.run` .

Still don't have solution. But it working for now

fulkast, 2018-01-17:
nppi was removed from CUDA in the transition 7.5 -> 8.0.
This post here says more:
https://devtalk.nvidia.com/default/topic/973151/no-nppgrabcut-on-cuda-8-0/

closed 2018-06-28 by Myzhar