stereolabs/zed-ros-wrapper #176: issue trying to use zed with jetson tx1

gal1017, 2018-01-11:
Hi,

We tried to run the Zed Explorer and we getting the Error:

QXcbConnection: XCB error: 145 (Unknown), sequence: 164, resource id: 0, major code: 139 (Unknown), minor code: 20
Segmentation fault
( We using MobaXterm for ssh-ing)
the same error is receiving also for zedFu, zed Depth view. zed calibration is fine ( we calibrated the camera...) .

**the Diagnostic output :

Running ZED SDK Components Diagnostic : OK

Running hardware Diagnostic : OK

Running cuda Diagnostic : OK

Running usb Diagnostic : OK

Running zed Diagnostic : Failed
Error:
Can't initialize the camera with all resolutions/fps. Check your USB connection.
I attached the log.**

some details about the setup:

16.04.3 LTS (GNU/Linux 4.4.38-tegra aarch64)
Cuda 8
ZED_SDK_Linux_JTX1_v2.2.1
can you help us please?

thanks,
Gal.

avirtuos, 2018-03-23:
I'm facing the same issue and also some challenges with the fact that the latest JetPack 3.2 uses CUDA 9.0 and Zed has only released an updated SDK for the TX2 w/CUDA 9.0 but not the TX1

nathangeorge1, 2018-07-23:
I'm also currently facing this issue; Jetpack 3.2 with CUDA 9 seg faults; I don't think this issue should be closed because the TX1 is still not working

nathangeorge1, 2018-07-23:
I got it working by switching my USB hub from the 2.0 version to the 3.0 version. Because it either seg faults or says the camera failed to open, it might be useful to catch these exceptions and mention that a 3.0 hub is required

Myzhar, 2018-07-23:
ZED uses USB3.0, so an USB3.0 hub is mandatory, even better if powered by an external source.
To check why the camera is not working you can use the "Diagnose Tool" available in the ZED tool folder.

nathangeorge1, 2018-07-23:
Most camera applications I've used catch seg faults and camera failures and warn the user of the need to use different hardware, like USB 3.0; any chance that could be implemented?

ravikt, 2018-08-03:
@Myzhar is correct. The camera will work well with USB 3.0 powered hub. The diagnostic test may still fail, but the camera will work.

closed 2018-11-24 by Myzhar