stereolabs/zed-ros-wrapper #178: Unable to open ZED prompt

InigoCastro, 2018-01-17:
Hello everyone,

I would like to tell you about an issue I am experiencing for which I haven not found a fix yet. I recently went through the process of setting up my computer to support the ZED camera ( installing SDK 2.2, openCV 3.1 and CUDA 8.0 in Ubuntu 16.04) as I recently changed my laptop. The issue starts when I install the ZED SDK and the message `ZED Detected, but cannot open Camera. Please check that you don't have another camera connected...` shows in the terminal. After this I tried to use the ZED tools that come as executables in the SDK installation. None of them were able to connect with the ZED camera, prompting different errors such as`Unable to open the ZED` (ZED Explorer). 

After reading a bit about similar issues I discovered that in early releases of the SDK the kernel version of Ubuntu caused problems. In case this is relevant, my kernel version is`4.13.0-26-generic`. Also I cheked if my new laptop maybe lacked USB 3.0 ports, but this is not the case.

At this point I am wondering if this is a hardware of software issue, since I changed my platform recently. However, there is no sign of anything being unusual from the laptop side. May it be that my ZED is not working properly?

Any help or advice on this issue would be appreciated.

adujardin, 2018-01-17:
Hi,

This is probably related to your kernel version indeed (https://github.com/stereolabs/zed-examples/issues/34). Either install ZED SDK 2.2.1 with CUDA 9 or stick with ZED SDK 2.2.0 with an older kernel.

InigoCastro, 2018-01-22:
Hello.

Thank you for your comment.

I indeed can open the ZED camera with ZED SDK 2.2.1 and CUDA 9.0. However, this generates another issue for me, as a complete install of CUDA 9.0 includes a NVIDIA accelerating graphics version that does not seem to be supported by my computer. I believe this since agreeing  (during the .run file installation) to install this driver leads to the infamous login loop error. If instead I do not install this driver I can open the ZED just fine, but it leads again to an error, which has already been discussed in [https://github.com/NVIDIA/nvidia-docker/issues/191](url) .

Therefore I have decided to go on with the older kernel. It would be useful for me if you could give me a hint of which version I should try. Maybe a 4.0XX ?

Thank you in advance.