stereolabs/zed-ros-wrapper #179: "Camera not detected" when I run the executable program in zed sdk.

wangjhit, 2018-01-22:
ZED SDK 2.1.2  worked fine a few days ago，but today I face a problem.
When I run the zed diagnostic program,no problem is detected.

> ZED Diagnostic Tool
Running ZED SDK Components Diagnostic : OK  
Running hardware Diagnostic : OK  
Running cuda Diagnostic : OK  
Running usb Diagnostic : OK  
Running zed Diagnostic : OK  
Diagnostic finished, a file has been generated : ZED_Diagnostic_log.json
Press any key...

but when I run some  program in bin folder or tools folder,It throw the message"camera not detedted"

> ./ZED_Camera_Control 
Error code:  Camera Not Detected

> ./ZED_Camera_Control 
Error code:  Camera Not Detected

why?

adujardin, 2018-01-22:
Hi,
Which Linux kernel version are you using? (`uname -r` command to find out)
Did you upgrade recently?

wangjhit, 2018-01-22:
Hello,
My kernel version is 4.13.0-26-generic.
However I am not sure whether I have upgraded it.

adujardin, 2018-01-23:
The kernel is the issue, it's not compatible with ZED SDK < 2.2.1. You can keep using 2.1.2 but you'll need to use an older kernel.
Related issues: https://github.com/stereolabs/zed-ros-wrapper/issues/178#issuecomment-358390801 and https://github.com/stereolabs/zed-examples/issues/34.

sinanmut, 2018-03-23:
Hi,

I have been using ZED camera for couple of weeks for some visual tests. Today it stopped responding and when I try to run "ZED Explorer" or other tools in the SDK. They don't respond and the camera gets quite hot after a while. When I try to run the samples in the  SDK, most of the time the output is "NO CAMERA DETECTED". I already updated the ZED SDK to the latest.

Could you please let me know why it may happen?

adujardin, 2018-03-23:
Hi @sinanmut,

This could be a hardware issue.
You should contact the support at support@stereolabs.com.

closed 2018-03-23 by adujardin

terzakig, 2018-03-23:
Hi,
Thanks a lot for the quick response. I already dropped an email to your support group. I thought it might be related to ubuntu kernel or recent nvidia/cuda updates. 

Thanks again. 
Best

sinanmut, 2018-03-23:
Hi,
Thanks a lot for the quick response. I already dropped an email to your support group. I thought it might be related to ubuntu kernel or recent nvidia/cuda updates.

Thanks again.
Best