stereolabs/zed-ros-wrapper #182: cuda not found

redone-technologies, 2018-02-03:
i installed cuda-9.1, zed-SDK 2.3.

-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - zed_wrapper
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'zed_wrapper'
-- ==> add_subdirectory(zed-ros-wrapper)
CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find CUDA: Found unsuitable version "7.5", but required is exact
  version "9.1" (found /usr)
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:386 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.5/Modules/FindCUDA.cmake:949 (find_package_handle_standard_args)
  /usr/local/zed/zed-config.cmake:42 (find_package)
  zed-ros-wrapper/CMakeLists.txt:22 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/hyun/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/hyun/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:290: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

HUDiwan, 2018-03-01:
I am having same problem, has anyone figured out why this issue exists, or how can i fix it?

WASCHMASCHINE, 2018-03-02:
Have you tried uninstalling CUDA 7.5?

>Could NOT find CUDA: Found unsuitable version "7.5", but required is exact
version "9.1" (found /usr)

trouble96, 2018-03-21:
Same problem and I tired after remove CUDA 7.5 but still same error.

wangdada123, 2018-03-21:
i have the same problem ,if someone fixed it ?

asimonov, 2018-03-26:
wow, seems to be common issues with ZED/CUDA.
with catkin -- did you run 'catkin_clean' before re-building? may have to as it may have cached cuda locations.

I have found a radical way to solve ZED+CUDA+ROS wrapper issues.
just follow NVIDIA instructions here: https://github.com/NVIDIA-Jetson/redtail/wiki/Skypad-Software-Setup-Guide#optional-zed-setup
this worked for me!

saesaem, 2018-04-16:
I have the same problem. I tried above two solution. But, it doesn't work for me. How to fix it?

wangdada123, 2018-04-16:
i solved this problem with cuda9.1 zed SDK 2.3. i guess some files didn't be removed completely  .when we update to the new zed SDK , it will conflict with the old version . if we didn't clean all and with zed-wrapper , it will link to the old zed SDK .

Myzhar, 2018-06-28:
When you change CUDA version on the machine you need to clean the cache of CMake otherwise it continues to see CUDA with the previous version.

The easiest way to do it is to remove "build" and "devel" folders in your catkin workspace and to start a new "catkin_make" from scratch

closed 2018-06-28 by Myzhar