stereolabs/zed-ros-wrapper #185: Invalid calibration file on ZED

JD-ETH, 2018-02-17:
Wrapper keeps downloading despite the calibration file seemingly already downloaded.
TX2, JetsonPack 3.2 and newest sdk . Compile successfully.

> auto-starting new master
> process[master]: started with pid [5515]
> ROS_MASTER_URI=http://localhost:11311
> 
> setting /run_id to 38d12584-13c7-11e8-ad06-00044b8d7ef0
> process[rosout-1]: started with pid [5528]
> started core service [/rosout]
> process[zed/zed_wrapper_node-2]: started with pid [5545]
> process[zed/zed_state_publisher-3]: started with pid [5546]
> [ INFO] [1518860701.758510748]: Initializing nodelet with 4 worker threads.
> [ INFO] [1518860701.817420736]: odometry_frame: map
> [ INFO] [1518860701.817495361]: base_frame: zed_center
> [ INFO] [1518860701.817558461]: camera_frame: zed_left_camera
> [ INFO] [1518860701.817605875]: depth_frame: zed_depth_camera
> [ INFO] [1518860701.817652520]: Publish map [TRUE]
> ZED (Init) >> Depth mode: PERFORMANCE
> ZED (Init) >> Video mode: HD720@60
> ZED (Init) >> No calibration file found. Downloading it... 
> ZED (Init) >> Invalid calibration file. Recalibrate your camera using ZED Calibration tool or download your calibration file on https://calib.stereolabs.com
> [ INFO] [1518860706.707777201]: Unknown
> ZED (Init) >> Depth mode: PERFORMANCE
> ZED (Init) >> Video mode: HD720@60
> ZED (Init) >> Wrong value in calibration file. Downloading it... 
> 

Diagnosis result seems fine, although could drop frame some time

> 
> ZED Diagnostic Tool
> 
> Running ZED SDK Diagnostic : OK  
> 
> Running Processor Diagnostic : OK  
>   valid:
>     Processor:   ARMv8 Processor rev 3 (v8l)
> 
> 
> Running Graphics Card Diagnostic : OK  
> 
> Running Camera Test Diagnostic : OK  
>   valid:
>     USB Bandwidth:  OK
>     Camera:  OK
> 
> 
> Running USB Diagnostic : OK  
> 
> 
> Diagnostic finished, a file has been generated : ZED_Diagnostic_Results.json
>

nesnes, 2018-02-21:
Your calibration file might be outdated.
As the calibration can't be used, the SDK will try to download the factory calibration, but as it is outdated, it might cycle around this error.

The solution is to open the Calibration tool from /usr/local/zed/tools and follow the instructions to recalibrate your ZED.

JD-ETH, 2018-02-23:
You are right. Thanks for the answer.

closed 2018-02-23 by JD-ETH