stereolabs/zed-ros-wrapper #198: zed_wrapper/ZEDWrapperNodelet issues

abanay, 2018-04-09:
i had my zed camera work well last week i had this error and ZED Depth Viewer doesn't run  

[ERROR] [1523275821.084599331]: Failed to load nodelet [/zed/zed_wrapper_node] of type [zed_wrapper/ZEDWrapperNodelet] even after refreshing the cache: Failed to load library /home/letsgeomat/catkin_ws/devel/lib//libZEDWrapper.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libnvidia-fatbinaryloader.so.387.26: cannot open shared object file: No such file or directory)
[ERROR] [1523275821.084643072]: The error before refreshing the cache was: Failed to load library /home/letsgeomat/catkin_ws/devel/lib//libZEDWrapper.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libnvidia-fatbinaryloader.so.387.26: cannot open shared object file: No such file or directory)

adujardin, 2018-04-09:
Hi,
If Depth Viewer doesn't work and since this appears (from the log) :
`libnvidia-fatbinaryloader.so.387.26: cannot open shared object file: No such file or directory`
It probably means it's an issue with your system, something like your Nvidia driver.
Try reinstalling it. The Cuda samples and ZED Depth Viewer should work before trying to launch the ROS wrapper.

abanay, 2018-04-09:
Reinstall cuda  ?? I am affraide to get some problems can you gelp me a
link or something  which can help me to good reinsta it

abanay, 2018-04-09:
i reinstall cuda and sdk but  zed viewer not working

abrandemuehl, 2018-04-09:
Have you tried rebuilding the zed-ros-wrapper in your catkin workspace?

abanay, 2018-04-09:
yeah but no result i will reinstall ubuntu :( :(  :(

maciejmatuszak, 2018-06-05:
I think there are compatibility issues between latest kernels 4.13 and nvidia drivers. I found post on nvidia site recommending nvidia-390 and later drivers for 4.13 onward kernels. I upgraded the nvidia drivers on its own but no luck. Then I used the cuda network deb installer to reinstall cuda 9.1 and it installs the latest nvidia drivers as well but this time all is working. 
Be careful the with network installer it will install latest cuda bu default which right now is 9.2. You will have to specify the version "sudo apt-get install cuda-9-1"

closed 2018-06-28 by Myzhar