stereolabs/zed-ros-wrapper #202: Compiling Error : Could NOT find CUDA: Found unsuitable version "9.0", but required is exact   version "9.1" -- Even though I'm sure I have CUDA 9.1 installed

ferhatm, 2018-04-20:
Here is the error while compiling: 
`-- +++ processing catkin package: 'zed_wrapper'
-- ==> add_subdirectory(rover_18/zed-ros-wrapper)
CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find CUDA: Found unsuitable version "9.0", but required is exact
  version "9.1" (found /usr/local/cuda-9.0)
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:386 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.5/Modules/FindCUDA.cmake:949 (find_package_handle_standard_args)
  /usr/local/zed/zed-config.cmake:42 (find_package)
  rover_18/zed-ros-wrapper/CMakeLists.txt:22 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/ferhat/rover_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/ferhat/rover_ws/build/CMakeFiles/CMakeError.log".
Makefile:2726: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
`

And this is my output for cuda version, nvcc -- version : 
`nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2017 NVIDIA Corporation
Built on Fri_Nov__3_21:07:56_CDT_2017
Cuda compilation tools, release 9.1, V9.1.85
`

Any idea ? 
Thanks

Librarvl, 2018-06-26:
you could try to recreate a ROS Workspace

Myzhar, 2018-06-28:
When you change CUDA version you must clean the cache of CMake. The easiest way is to delete "build" and "devel" folders in your catkin workspace

closed 2018-06-28 by Myzhar