stereolabs/zed-ros-wrapper #211: Depth computations for external stereo pair

pavloblindnology, 2018-06-08:
Is it possible to feed ZED SDK with external stereo pair,
the same way it's done for svo files, which, AFAIU, store only stereo pair without any depth map?
Thanks.

adujardin, 2018-06-28:
The ZED SDK is a driver for the ZED cameras and is therefore strongly tied to the ZED. 
It doesn't support third-party cameras by design.
Sorry.

closed 2018-06-28 by adujardin

pavloblindnology, 2018-07-02:
But still, ZED SDK/camera can be fed with SVO file which doesn't contain depth info (am I right?).
Can I create SVO file from external stereo pair?
My intention is to playback ZED depths from ZED stereo pairs recorded to rosbag. And I don't want to store depth to rosbag to save storage space and to have the ability to play with depth generation settings while playing back.
Thanks.