stereolabs/zed-ros-wrapper #219: Garbing image from ZED fails with SDK 2.1.2 and CUDA 8

ebimor, 2018-06-25:
Hello,

I am trying to use ZED camera with CUDA 8 on a system with Linux 16.04 and I only need the RGB color images in HD mode (no depth calculation or ...). 

- Using opencv cv::VideoCapture, I can read the images and everything works fine. However, it takes a lot of CPU usage (40 percent, on a system with 16GB RAM) which affects other tasks that I am doing on my machine such as SLAM, control, etc.

- I tried to do the same thing with ZED SDK and I used your example code to open the camera and retrieve the left and right images. Note that since I am using CUDA 8, I had to use SDK 2.1.2. However, it fails at opening the camera.

So I have two questions:
- If I use ZED SDK, should I expect to see less CPU usage for grabbing images from the camera (compared to openCV)? 
- If yes, do you have an SDK that would work with CUDA 8 on Linux 16.04?

Thanks,

obraun-sl, 2018-06-25:
Can you post a lsusb output when the camera is plugged in?

Myzhar, 2018-09-08:
@ebimor since this is not an issue related to ROS you can contact the support if you need further help on it: support at stereolabs.com

closed 2018-09-08 by Myzhar