stereolabs/zed-ros-wrapper #220: ZED (Init) >> WARNING: FPS is too low to enable positional tracking and spatial mapping. Consider using PERFORMANCE parameters

Librarvl, 2018-06-26:
Hello,
I want to try running zed under ros. But when I was using rviz, I found this error after choosing PointCloud2. And it's not running smoothly.

So I want to find the the PERFORMANCE parameter and let it run smoothly. Or maybe some ideas will be better. Please teach me.Thanks.

Myzhar, 2018-06-28:
Can you please give information about Operating System, Elaboration Device and GPU?

Librarvl, 2018-06-30:
These are the configuration of my computer:
- Operating:Ubuntu 16.04
- CPU:AMD Athlon(tm) X4 750 Quad Core Processor × 4 
- memory:8G
- GTX1070Ti   CUDA 9.0

And the last, your suggestion is valid. 
Thank you very much.

adityajain07, 2018-07-02:
Try reducing the video resolution to HD720 and increase the fps to 60 in zed_camera.launch file. In ZED parameters column, resolution = 2 and frame_rate = 60. Should solve your issue.

sophie-shin, 2018-08-17:
Same problem with Matlab. 

FPS is too low to enable positional tracking and spatial mapping. Consider using PERFORMANCE parameters. 
-- with resoultion HD720 and 60 Fps

Myzhar, 2018-09-08:
I close the issue but you can continue to post comments if you need further help

closed 2018-09-08 by Myzhar