stereolabs/zed-ros-wrapper #222: What is the best SLAM algorithm for ZED on ROS

ArtlyStyles, 2018-06-27:
Hi, I have built a MIT Racecar platform https://mit-racecar.github.io. Currently, I do not have the Hokuyo lidar. I am wondering if I can run SLAM algorithm with ZED. Which algorithm is the best for ZED?

Thanks,

rbonghi, 2018-06-28:
Hi @ArtlyStyles , I used on my robot [Panther](https://rnext.it/panther) different type of algorithms:
* [gmapping](http://wiki.ros.org/gmapping)
* [rtabmap](http://wiki.ros.org/rtabmap_ros)
* [Cartographer](https://google-cartographer-ros.readthedocs.io/en/latest/)
* [hector slam](http://wiki.ros.org/hector_slam)

My first approach was use the [pointcloud to laserscan](http://wiki.ros.org/pointcloud_to_laserscan) converter and use the laserscan message to build my map. More or less your same case with your racecar, however the [rtabmap](http://wiki.ros.org/rtabmap_ros) with the information from the ZED can build a 3D map.

You can see one of my reconstructions here https://skfb.ly/6sUS7

ravikt, 2018-08-03:
I have used ORBSLAM2 and SVO. Both seems to work well with ZED camera.

Myzhar, 2018-09-08:
I close the issue but you can continue to pay comments if you need further help

closed 2018-09-08 by Myzhar

belal-ibrahim, 2021-01-15:
@rbonghi May i ask about the results of gmapping using zed without laser sensor? 
Were there any drawbacks.
Can zed along with odom be good in mapping large scale env?

Myzhar, 2021-01-16:
@belal-ibrahim `gmapping` requires a laser scan message to work, so you must convert a 3D point cloud (or the depth map) to a simulated 2D scan message losing a lot of useful information.
I suggest you to take a 3D approach using the 3D mapping feature provided by the ZED SDK or using RTABmap that is the state of the art package for 3D localization and mapping