stereolabs/zed-ros-wrapper #224: Exposure ZED Wrapper 2.0.x

getaleks, 2018-07-02:
Hi,

I am using the ZED Wrapper version 2.0.x. I would like to change the exposure settings (turn off auto exposure and set a manual value).

Unfortunately, the launch file does not offer an exposure parameter. Is there another way to set the parameter?

Myzhar, 2018-07-02:
The wrapper 2.0.x does not have a configuration param for exposure.

You can set it by yourself using the "setCameraSettings" parameter from ZED SDK:
https://www.stereolabs.com/developers/documentation/API/v2.4.0/classsl_1_1Camera.html#abf8a74d57c651ab02843be7e9f5c269a

otherwise you must checkout the latest "master" branch that exposes dynamic reconfigure for confidence, exposure and gain.

getaleks, 2018-07-02:
Hi Myzhar,

thanks for the quick response!

Unfortunately, updating to the latest wrapper is not an option for me because, as I understood it, the newest SDK requires CUDA 9.0. Updating the machine to CUDA 9.0 is not an option for me at the moment. Also, we have made certain custom changes to the wrapper for our specific application and do not wish to add these changes again.

Do you have some sample code how and where to set the exposure parameter in the ZED Wrapper 2.0.x?

Thanks!

Myzhar, 2018-07-02:
Yes, you can watch how they are set here:
https://github.com/stereolabs/zed-ros-wrapper/blob/master/src/zed_wrapper_nodelet.cpp#L633-L648

getaleks, 2018-07-02:
Hi Myzhar,

thanks for the link.

I tried implementing analogously to your code sample.

Are you sure the method zed.setCameraSettings is available in earlier SDKs?

During build I'm getting the error: 
```
class std::unique_ptr<sl::Camera> has no member named 'setCameraSettings' zed.setCameraSettings(sl::CAMERA_SETTINGS_EXPOSURE, 0, true);
```

obraun-sl, 2018-07-02:
Hi,

Before 2.0, this function was called `setCameraSettingsValue` with the same function parameters.

getaleks, 2018-07-02:
Hi,

thanks for the tip.

It is actually zed->setCameraSettings in my version.

It works now:)

Thanks again for the help everybody!

closed 2018-07-02 by Myzhar

tingxueronghua, 2019-01-24:
excuse me, I have a problem about the exposure. Is the exposure really set as we do in the 'setCameraSettings'? I set the exposure but when we move the camera the exposure seems to change.

Myzhar, 2019-01-24:
@tingxueronghua have you set to false before changing exposure value?

From documentation:
```
auto_exposure: Enable/Disable auto control of exposure and gain
exposure: Exposure value when manual controlled (auto_exposure=0)
```
https://www.stereolabs.com/docs/ros/zed_node/#dynamic-parameters

PS Please do not reply to a closed issue, open a new issue instead