stereolabs/zed-ros-wrapper #225: libZEDWrapper.so: undefined symbol: _ZN2cv3Mat20updateContinuityFlagEv

claydergc, 2018-07-02:
Hello,

I'm getting this error when trying to execute the command 'rostopic echo' from left and right color raw images from the zed camera. It seems to be an error with finding a specific function in the OpenCV libraries. I am working with Ubuntu 18.04, ROS melodic and installed Opencv 4.0.

Which version of OpenCV should I have installed to get the nodes working correctly?

Thanks.

Myzhar, 2018-07-02:
The node has been tested with ROS Kinetic and OpenCV 3. 
OpenCV4 has not yet been released officially so support for that version is not guaranteed

claydergc, 2018-07-02:
Thanks! problem solved

closed 2018-07-02 by claydergc

dev-kasibhatla, 2018-08-30:
Hello
is this going to work with ubuntu 18 + ros melodic + opencv 3.4?

thanks

Myzhar, 2018-08-30:
The ZED wrapper has been currently tested under Ubuntu 16.04 + ROS Kinetic
You can try to compile the Wrapper under Ubuntu 18 for ROS Melodic, but be sure that the CUDA version you installed on your system is the version supported by the ZED SDK you installed.

horitahiroshi, 2018-09-27:
> Thanks! problem solved

Hi, @claydergc .
To solve this problem you've just changed the OpenCV, downgrading to OpenCV3?
So it worked fine with Ubuntu 18.04 and ROS melodic, right?

Thanks.

claydergc, 2018-09-27:
Yes, that is what I did.