stereolabs/zed-ros-wrapper #227: Low FPS when subscribing to point cloud topic.

RhysMcK, 2018-07-04:
I have notice that the FPS from the zed-ros-wrapper drops from 15 fps ( as desired in launch file) to about 5 fps when i subscribe to the point_cloud topic. Obviously the additional computational overhead when having to calculate the point cloud slows down the main working thread. Has there been any work on computer the point cloud asynchronously? 

Cheers.

Myzhar, 2018-07-04:
I need more information to help you:
* platform (CPU, GPU, ...)
* SDK version

One of the main cause maybe running Rviz on the same machine.
To reduce the power requested by Rviz you could try to change this settings:
- Unreliable: ON
- Style: Points
- Size (Pixels): 1
- Queue size: 1

RhysMcK, 2018-07-04:
Thanks for the response. This is running with GPU on version SDK 2.4.  Those values I stated above were determined via the `rostopic hz`. I am only running `zed.launch`

Myzhar, 2018-07-04:
Which GPU and CPU?

RhysMcK, 2018-07-04:
Sorry, i edited after inital response.

Myzhar, 2018-07-04:
Ok, I think you can take advantage of the new release of the wrapper that will be released with new SDK in the next days

Russ76, 2018-07-05:
Does using the notelet instead of the node help with the fps issue?

RhysMcK, 2018-07-05:
Hi @Russ76, i am currently already using this as a nodelet.

RhysMcK, 2018-07-05:
I've done a little bit of investigation, and the bottle neck appears in the `zed_wrapper_nodelet.cpp`:lines 708- 716:
```
if (cloud_SubNumber > 0) {
     zed.retrieveMeasure(cloud, sl::MEASURE_XYZBGRA);
     point_cloud_frame_id = cloud_frame_id;
     point_cloud_time = t;
     publishPointCloud(width, height, pub_cloud);
}
```
it takes about 0.17 seconds to complete this block, which is about 5-6hz. About 70% of that time is cosumed in the `publishPointCloud()' call.

Myzhar, 2018-07-05:
Thank you for the investigation @RhysMcK 
`publishPointCloud` is the function that have been improved in the version that we are going to release

Myzhar, 2018-08-06:
Performances improved in 1b16546e92677b4e5d9ff6688d49eb1812e10f3d

closed 2018-08-06 by Myzhar