stereolabs/zed-ros-wrapper #229: Could NOT find CUDA: Found unsuitable version "7.5", but required is exact   version "9.1" (found /usr)

flightlesskite, 2018-07-04:
`CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find CUDA: Found unsuitable version "7.5", but required is exact
  version "9.1" (found /usr)
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:386 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.5/Modules/FindCUDA.cmake:949 (find_package_handle_standard_args)
  /usr/local/zed/zed-config.cmake:42 (find_package)
  zed_wrapper/CMakeLists.txt:22 (find_package)
`

I am getting this issue when attempting to install the ZED wrapper for a bit now...
However, I have installed CUDA 9.1 and don't know why it's saying I'm using 7.5...
Managed to install the ZED sdk with CUDA 9.1 alright so I am not sure why this is an issue...

Any idea how to resolve this?

Myzhar, 2018-07-04:
If you have installed CUDA 7.5 before the problem is with the cache of CMake.
To be sure to clean it you can delete the folders `build` and `devel` in your catkin workspace and then run a new `catkin_make` command

Myzhar, 2018-07-04:
What is the reply to the command 
`$ nvcc -V` ?

closed 2018-09-08 by Myzhar

vonunwerth, 2019-09-09:
Removal of `build` and `devel` folders worked for me.

camzilla1050, 2021-01-24:
@vonunwerth tnx, just cmake ..   it works for me