stereolabs/zed-ros-wrapper #231: Running ROS and sl::Camera

Thayjes, 2018-07-10:
Hi,
I was wondering if it was possible to open an sl::Camera object as well as run zed ros at the same time somehow?

nesnes, 2018-07-10:
Hi, 
The ZED can be used by only one program at a time. However, if you have 2 ZED, you can have one running in the ros wrapper and one in a custom program (using the SDK) at the same time.

As your question was probably more about the same ZED, the answer is no. The second program will have to subscribe to the ROS topics of the ZED to get images and depth from it.

Can you describe your use-case to find why you needed that?

Thayjes, 2018-07-10:
Thanks for your reply. Yes that is what I thought. You are right, I am talking about the same ZED. The reason for doing this is, I want to the timestamp when the image was actually captured (both depth and rectified) and not the timestamp that ROS gives me when I subscribe to the ROS Node. I read somewhere that this can be done only be using the gettimestamp function (sl::Camera.getTimestamp(sl::TIME_REFERENCE::TIME_REFERENCE_IMAGE)). But to do this, I need to create an sl::Camera object and open it. And also grab. But this is not possible if the ROS wrapper is already running? Is there anyway to get this information within ROS itself and not needing to create a sl::Camera object?

nesnes, 2018-07-10:
@Myzhar Can give you more info, but he's about to publish an update where the timestamp in the ROS messages will correspond to the getTimestamp() call from the SDK. So writing a simple subscriber will solve your issue.
https://github.com/stereolabs/zed-ros-wrapper/blob/devel/src/nodelet/src/zed_wrapper_nodelet.cpp#L1100

Thayjes, 2018-07-10:
I see, thanks for that information. I am actually running this on Jetson TX1. And I believe I am running an older SDK (2.0). What version will I need to upgrade to, to make use of this update. Also any idea when this will be published? 
The idea which I was thinking before, was to write my own node that publishes the depth and rectified images with the corrected timestamp. And then subscribe to these custom topics. But it would be much less work if this update was going to happen soon. 
Thanks again for your help.

nesnes, 2018-07-10:
This update targets the upcoming ZED SDK 2.5 which is compatible with the TX1 using JetPack 3.2 (CUDA9).
I'll recommend you to update your Jetson board, as the SDK and the wrapper should be published in a few days.

If you don't want to, you can also copy the timestamp [line](https://github.com/stereolabs/zed-ros-wrapper/blob/devel/src/nodelet/src/zed_wrapper_nodelet.cpp#L1100) and the [function](https://github.com/stereolabs/zed-ros-wrapper/blob/devel/src/tools/src/sl_tools.cpp#L161) it uses to update your own wrapper version.

Thayjes, 2018-07-10:
Thanks for your clear explanation. Just to clarify, with these changes in the code, the timestamp that ZED ROS wrapper uses and the SDK will be the same, i.e the time at which the image was extracted from the USB stream?

Thayjes, 2018-07-10:
I tried updating my software on the TX1 to ZED 2.3.3. And I got the following error after launching the zed wrapper. Any idea why I might have got this error?
[image]

Myzhar, 2018-07-12:
@Thayjes I confirm that the new upcoming version will give the timestamp in SDK time reference and not in ROS time. You can verify it here:
https://github.com/stereolabs/zed-ros-wrapper/blob/devel/src/nodelet/src/zed_wrapper_nodelet.cpp#L1085
where `slTime2Ros` is a new function to convert StereoLabs timestamps to ROS time:
https://github.com/stereolabs/zed-ros-wrapper/blob/devel/src/tools/src/sl_tools.cpp#L161-L165

closed 2018-07-12 by Myzhar