stereolabs/zed-ros-wrapper #233: Missing tf

Thayjes, 2018-07-10:
Hi,
I am running ZED SDK 2.3.3 on Jetson TX1 Linux. Once I run my launch file, I get the following message:
[image]
Why does it say the tf is not available? Do I need to download this or specify something in the launch file?
Will it cause any issues? Thanks in advance.

Myzhar, 2018-07-16:
Please update the SDK to the latest available for Jetson TX1 and let me know if the warning persists

closed 2018-08-06 by Myzhar

xiaodongzhao, 2019-03-06:
Hi @Myzhar , 
I am using the latest v2.7 SDK on Ubuntu 18.04 and also having this warning about "zed_left_camera_frame" missing (from roslaunch zed_wrapper zed.launch). Is there a fix to this warning? 

thanks.

Myzhar, 2019-03-06:
@xiaodongzhao do you get the warning continuously or only once at start?

xiaodongzhao, 2019-03-06:
Hi @Myzhar, it only appears once at the start.

Myzhar, 2019-03-06:
Ok, so you can ignore it. It appears because the TFs published by the Robot State Publisher node are not yet propagated.
There is a fix for that warning that will be added in the next versions of the wrapper.

xiaodongzhao, 2019-03-06:
Thanks very much for your reply @Myzhar