stereolabs/zed-ros-wrapper #234: tf tree has two root node

ArtlyStyles, 2018-07-15:
Hi all,

I am building a MIT-RACECAR with ZED stereo camera and Yokuyo LiDar on it. When I visualized the /tf, it showed two trees:

One tree has 1)odom as root, then 2)base_link as second layer and then 3)Laser, Base_footprint, IMU as children of base_link

Another tree has 1)Map as root, then 2)ZED_center as second layer and then 3)ZED left camera and ZED right camera as children of ZED_Center.

I remember that tf can have only one tree, one root node. Then how can I combine the two trees into one?

Myzhar, 2018-07-16:
Probably you executed the ZED node in two different ways:
1) you launched it using "zed.launch" and `roslaunch` (`map` as root)
2) you launched it without using `roslaunch` but using `rosrun` directly with the `zed_wrapper_node` executable (`odom` as root)
The better way is to use `roslaunch`, to be more precise the command is
`roslaunch zed_wrapper zed.launch`

ArtlyStyles, 2018-07-16:
I did used two launch files. One is for the Car, which starts the driving node, IMU, laser, etc. The other one is zed.launch as you described.  
I just want to merge the two trees into one tf tree. But I do not know how to do it.

Myzhar, 2018-07-17:
I do not know the TF tree of the Racecar, but the standard way of work when you have a sensor is to provide a static trasform (in the URDF or in the code) from one of the link of the "robot" (the car in your case) and the root of the sensor.

ArtlyStyles, 2018-07-17:
@Myzhar thanks. So I need to modify URDF to provide a transform from my the robot base_link to ZED zed_center? I think the transform should be based on the physical location of where the ZED is on the car. Correct?

Myzhar, 2018-07-17:
Exactly

ArtlyStyles, 2018-07-17:
I could not find the URDF under my cars workspace. Do you know where the URDF is saved?

Myzhar, 2018-07-17:
Usually it is in the URDF folder. If it does not exist you must create a new one for the Racecar. But this is not related to zed-ros-wrapper, it's a ROS issue.
You can find more information on http://answers.ros.org

closed 2018-08-01 by Myzhar